openwrt/luci #2132: Dhcp static leases - IPv4 Address field too small

mbo2o, 2018-09-05:
While editing/browsing static leases on the Network→DHCP and DNS page all the IPv4 address are obscured by the pull-down menu button. While editing one of the address entries the column becomes wider and all entries are readable. The lease Time Field is wider than required. 
[image]

A possible work around would be too increase the size of the table header by adding extra text.
replace

/usr/lib/lua/luci/model/cbi/admin_network/dhcp.lua:308
`ip = s:option(Value, "ip", translate("<abbr title=\"Internet Protocol Version 4\">IPv4</abbr>-Address"))`
with
`ip = s:option(Value, "ip", translate("<abbr title=\"Internet Protocol Version 4\">IPv4</abbr>-Address xxx.xxx.xxx.xxx"))`

dibdot, 2018-09-05:
I don't see any problem here. We're using dynamic, mostly responsive templates in LuCI - without any hardcoded table structure/measurements. During editing you'll get/see always the full content (same applies to other fields e.g. like MAC address), after that you'll see the full content or a part of it - depends on (mobile) device, screen resolution, browser window/tab size etc. ... in my opinion it works as intended. 

Your proposed "fix" doesn't change that at all.

mbo2o, 2018-09-09:
It doesn't work as intended, the last octet is always hidden by the button of the combo-box.
Inspecting the element in the browser shows the width is set to 100%
```
.table.cbi-section-table textarea, .table.cbi-section-table select {
    width: 100%;
}
```
Setting it to auto will resolve the issue however I am not sure where to make this change.
```
.table.cbi-section-table textarea, .table.cbi-section-table select {
    width: auto;
}
```