openwrt/luci #2138: Would luci-app-statistics support collectd-mod-curl?

jinchizhong, 2018-09-09:
Hi everyone,

I'm interested in run collectd-mod-curl on OpenWRT.
I manually configured collectd-mod-curl, it works, and .rrd file appear on where I want.
But result doesn't appear on luci/Statistics/Graph.

After some research, I find it out, luci-app-statistics doesn't support collectd-mod-curl by now.
Is there any plan to support it?

If no, is it easy to implement? Can anybody guide me how to implement such a feature?

Thanks to all

hnyman, 2018-09-09:
> luci-app-statistics doesn't support collectd-mod-curl by now.
> Is there any plan to support it?

No.
You are the first one that has asked for it.

> If no, is it easy to implement? Can anybody guide me how to implement such a feature?

Rather easy.
You just do similar edits as for e.g. cpufreq and apcups module:
https://github.com/openwrt/luci/commit/df3c8728b995026a6df13efaf9a782b486de9ec5#diff-64c154d89ea097b12be97c2bc71b09da
https://github.com/openwrt/luci/commit/808f2617931dcd6f84637e1bff3a737b9368917c#diff-64c154d89ea097b12be97c2bc71b09da

You can do the editing in a live router, and once you get things working, you can then transfer your modifications to the git repo and submit a pull request here, to get your code merged into the official repo.

jinchizhong, 2018-09-10:
@hnyman Thank you,
I'll have a try next weekend.

jinchizhong, 2018-10-03:
Hi @hnyman 

I met a small problem, can you have a look?
[image]

In the first red box, I hope there should be '40 ms'. （the unit should be 'ms' not 'm')
And in the second red box, I hope there should be '30 ms'. (the value should x1000)

I can't find where to change them.

Below is my code by now.
```
-- statistics/rrdtool/definitions/curl.lua
module("luci.statistics.rrdtool.definitions.curl", package.seeall)

function rrdargs( graph, plugin, plugin_instance, dtype )
        return {
                title = "%H: cUrl Response Time for #%pi",
                y_min = "0",
                alt_autoscale_max = true,
                vlabel = "Response Time",
                number_format = "%5.1lfms",
                data = {
                        types = { "response_time" },
                        options = {
                                response_time = {
                                        title = ""
                                }
                        }
                }
        }
end
```

Thanks a lot

hnyman, 2018-10-04:
You are likely looking for "%5.1lf%Ss" or something similar.

https://oss.oetiker.ch/rrdtool/doc/rrdgraph_graph.en.html#PRINT

> %s
>     place this after %le, %lf or %lg. This will be replaced by the appropriate SI magnitude unit and the value will be scaled accordingly (123456 -> 123.456 k).
> 
> %S
>     is similar to %s. It does, however, use a previously defined magnitude unit. If there is no such unit yet, it tries to define one (just like %s) unless the value is zero, in which case the magnitude unit stays undefined. Thus, formatter strings using %S and no %s will all use the same magnitude unit except for zero values.

jinchizhong, 2018-10-08:
Hi @hnyman 

Thank you for your help, I just send a PR.
I'm not familiar with OpenWRT development environment.
So I edit them in my router, and then copy it to repo.
I hope I do not miss something.

Please have a look.
Thanks

feckert, 2019-04-23:
This was merged so I think we could close this issue / feature request

closed 2019-04-23 by feckert