openwrt/luci #2146: luci-app-openvpn unable to create new instances

davidc502, 2018-09-13:
luci-app-openvpn unable to create new instances.
LuCI Master (git-18.250.48934-18b4483) / Lede SNAPSHOT r8018-42f158314e

Users are reporting they are unable to save new Openvpn instances on build r8018  I've have also been able to duplicate and confirm this behavior.

dibdot, 2018-09-13:
The LuCI app itself was not changed, probably related to this core change where default config options have been removed: 
(https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=89b8ba96b404d2e412d5573e25348287255671dd )

dibdot, 2018-09-14:
@davidc502 My first assumption was wrong, please test proposed minimal patch - thanks.

davidc502, 2018-09-16:
Testing complete -- Added to parenthesis to line 62, and was able to create a new instance. 

Thank you

dibdot, 2018-09-16:
Great, thanks for re-testing!

closed 2018-09-16 by dibdot