openwrt/luci #2147: "Active DHCPv6 Leases" on status page is empty if system is running dnsmasq-full

jebrossard, 2018-09-13:
I see DHCPv6 leases displayed until I remove “dnsmasq” and install “dnsmasq-full”.  Otherwise the system is working fine, I can see in DHCPv6 leases being issued.

This wasn't an issue on any of the 17 versions but has been an issue on the 18 versions.

Regards,

Jim

jow-, 2018-09-18:
Please provide your `/etc/config/dhcp` and a scrubbed example of `/tmp/dhcp.leases` from your setup

jebrossard, 2018-09-24:
It seems I'm lying, I don't see any DHCPv6 leases.  But I'm sure I see DHCPv6 leases being issued in the log.

dhcp:
config dnsmasq
	option domainneeded '1'
	option localise_queries '1'
	option rebind_protection '1'
	option rebind_localhost '1'
	option local '/lan/'
	option domain 'lan'
	option expandhosts '1'
	option authoritative '1'
	option readethers '1'
	option leasefile '/tmp/dhcp.leases'
	option resolvfile '/tmp/resolv.conf.auto'
	option nonwildcard '1'
	option localservice '1'
	option dnssec '1'
	option dnsseccheckunsigned '1'

config dhcp 'lan'
	option interface 'lan'
	option start '100'
	option limit '150'
	option leasetime '12h'
	option ra 'server'
	option dhcpv6 'server'
	option ra_management '1'
	list dhcp_option '252,"\n"'

config dhcp 'wan'
	option interface 'wan'
	option ignore '1'

config dhcp 'guest'
	option start '100'
	option leasetime '12h'
	option limit '150'
	option interface 'guest'
	option ra 'server'
	option dhcpv6 'server'
	option ra_management '1'

dhcp.leases:
1537839170 e0:4f:43:d9:e6:d7 192.168.3.186  *
1537868050 c0:c1:c0:4b:ca:b9 192.168.1.103 *
1537850779 04:5d:4b:49:16:32 192.168.1.171 * 01:04:5d:4b:49:16:32
1537850651 00:18:39:43:3b:49 192.168.1.104 01:00:18:39:43:3b:49
1537871268 00:e0:4c:36:3c:73 192.168.1.127 *
1537868184 00:e0:4c:36:e9:17 192.168.1.112  *
1537866076 8c:3b:ad:e3:3c:f8 192.168.1.107 01:8c:3b:ad:e3:3c:f8
1537867042 c4:1c:ff:df:2a:6b 192.168.1.227 *
1537857704 00:26:2d:00:84:5b 192.168.1.100 01:00:26:2d:00:84:5b
1537856699 70:5a:b6:a7:51:9f 192.168.1.230 3 01:70:5a:b6:a7:51:9f
1537854386 bc:ae:c5:0f:f2:b3 192.168.1.102 5 01:bc:ae:c5:0f:f2:b3
1537854905 4c:60:de:5d:8e:e3 192.168.1.116 * *
1537864168 b8:27:eb:e6:08:9b 192.168.1.106 *
1537869657 78:d2:94:4f:ec:21 192.168.1.126 * *
1537871020 a0:6a:44:52:89:03 192.168.1.118 * 01:a0:6a:44:52:89:03
1537853529 ac:22:0b:e2:8d:8b 192.168.1.229 -cca1d8d466379174 01:ac:22:0b:e2:8d:8b
1537861626 0c:b2:b7:45:6c:b4 192.168.3.172 -0B3E 01:0c:b2:b7:45:6c:b4
1537832014 b4:f7:a1:be:e3:00 192.168.3.115 * 01:b4:f7:a1:be:e3:00
1537853321 18:b4:30:a8:a6:df 192.168.3.245 * *
1537871330 64:51:06:69:ba:09 192.168.1.101 01:64:51:06:69:ba:09
1537839435 d0:13:fd:64:d9:db 192.168.1.114 * 01:d0:13:fd:64:d9:db
1537853508 18:b4:30:be:9a:b0 192.168.3.120 01:18:b4:30:be:9a:b0

jebrossard, 2018-09-24:
I don't see this lease in the above:

Mon Sep 24 09:06:59 2018 daemon.notice odhcpd[1529]: Got DHCPv6 request
Mon Sep 24 09:06:59 2018 daemon.warn odhcpd[1529]: DHCPV6 RENEW IA_NA from 00010001232af947bcaec50ff2b3 on br-lan: ok 2601:601:9280:2081::102/128

jebrossard, 2018-09-24:
Yup, removing odhcpd-ipv6only has corrected the issue.  I just didn't have to do this with LEDE V17.  Thank you for leading me to discover the solution!

closed 2018-09-24 by jebrossard