openwrt/luci #2151: Recursion Protocol is generating wrong configuration

Jonny007-MKD, 2018-09-14:
Hi,

many thanks for this unbound UI, it makes it much easier to start with unbound. With my special network situation I noticed an error: The "Recursion Protocol" is not doing what it shall do.

### My network
I only have IPv4 internet access, but also local IPv6 addresses. Thus, I want unbound to be accessable via IPv4v (`do-ipv4: yes, interface 0.0.0.0`) and IPv6 (`do-ipv6: yes, interface: ::0`).

Because I cannot access upstream IPv6 DNS servers, I set "Recursion Protocol" to "IPv4 only". But, this only generates the directive `do-ipv6: no` which also disables the listening IPv6 socket.
Correct would be to not generate the `outgoing-interface ::0` directive.

dibdot, 2018-09-14:
@EricLuehrsen

dibdot, 2018-09-16:
fixed in trunk

closed 2018-09-16 by dibdot