openwrt/luci #2156: Luci should not generate new package releases with no change

luizluca, 2018-09-17:
Hello,

Currently, when any commit is made to a stable branch, all luci packages gets a new release, even those without changes. For those that install all "opkg list-upgradable" packages, there are some more precious kbytes consumed in /overlay. Since 18.06.1, there is already a new luci release. All luci-* packages in standard rom image have a an update but not real file change.

Would it be possible to have package versions be more conservative?

Could it use a fixed git-commit (a tag or a Makefile var) as a base commit when there is a openwrt release and use PKG_RELEASE as the "1+(number of commits since the base commit)". Each module/app/theme/(...) has its own directory and it would be simple to count the number of commits since the "fixed git-commit" inside each directories.

If there is a change that must recompile all packages, simply update the base git-commit. It will also automatically reset all "releases" back to 1.

luizluca, 2018-09-17:
I did a simple POC here https://github.com/luizluca/luci/commit/e3100dfd47968c3a4f94306dc727182a9d50053a

The result packages is this:

- **luci-app-adblock_git-18.260.69651-d5c5f90-2_all.ipk**
- luci-app-firewall_git-18.228.31946-f64b152-0_all.ipk
- luci-base_git-18.228.31946-f64b152-0_x86_64.ipk
- luci_git-18.228.31946-f64b152-0_all.ipk
- luci-lib-ip_git-18.228.31946-f64b152-0_x86_64.ipk
- luci-lib-jsonc_git-18.228.31946-f64b152-0_x86_64.ipk
- luci-lib-nixio_git-18.228.31946-f64b152-0_x86_64.ipk
- luci-mod-admin-full_git-18.228.31946-f64b152-0_x86_64.ipk
- luci-proto-ipv6_git-18.228.31946-f64b152-0_all.ipk
- luci-proto-ppp_git-18.228.31946-f64b152-0_all.ipk
- luci-theme-bootstrap_git-18.228.31946-f64b152-0_all.ipk

Note that luci-app-adblock is the only package that changed (2 commits ahead f64b152).
If release==0 is a problem, It could be +1.

The use of f64b152 is just an example of how packages would look like if adopted since 18.06 release. However, it cannot be used anymore as a base commit because there is already a newer version published (9541751).

jow-, 2018-09-18:
I'm neither a fan of tags or fixed commit IDs, we need to come up with another solution. Maybe something based off the ancestor branch

luizluca, 2018-09-19:
@jow- , the common ancestor (release vs master) will give you a base commit and it avoid new package releases on any commit. However, it introduces a new problem for commits that, even not touching submodules code, will affect all generated packages (like changing luci.mk)

The problem is to automatically detect when a commit requires or not new releases for all packages. It is normally a human decision. It is equivalent to bumping PKG_RELEASE on openwrt-packages.

How about using a new LUCI_RELEASE:=1 that sums with "number of commits since the base commit"? When a commit needs a full release, just bump LUCI_RELEASE.

Back to using "the common ancestor", it will work only for branches of master. "master" would still use the current behavior.

I'll update the POC

luizluca, 2018-09-19:
New POC using common ancestor and LUCI_RELEASE at https://github.com/luizluca/luci/commit/64dbdc7cb9f73fd280b33507a4316ecb6df6932b :
The result:

- luci-app-firewall_git-18.194.53747-b38306d-8_all.ipk
- luci-base_git-18.194.53747-b38306d-50_x86_64.ipk
- luci_git-18.194.53747-b38306d-2_all.ipk
- luci-lib-ip_git-18.194.53747-b38306d-1_x86_64.ipk
- luci-lib-jsonc_git-18.194.53747-b38306d-1_x86_64.ipk
- luci-lib-nixio_git-18.194.53747-b38306d-1_x86_64.ipk
- luci-mod-admin-full_git-18.194.53747-b38306d-19_x86_64.ipk
- luci-proto-ipv6_git-18.194.53747-b38306d-2_all.ipk
- luci-proto-ppp_git-18.194.53747-b38306d-1_all.ipk
- luci-theme-bootstrap_git-18.194.53747-b38306d-8_all.ipk

luizluca, 2018-09-23:
@jow- , should I PR this?

jow-, 2020-07-13:
This has been reworked in master

closed 2020-07-13 by jow-