openwrt/luci #2161: No IPv6-address on the status page

vgaetera, 2018-09-18:
- Network > Interfaces > WAN6 > IPv6 - Correct address here
- Status > Overview > Network > IPv6 Upstream > Address - No address, `::`

Packages version: `luci - git-18.247.71242-9541751-1`

jow-, 2018-09-18:
Please provide the output of `ifstatus wan6`

vgaetera, 2018-09-18:
```
# ifstatus wan6
{
        "up": true,
        "pending": false,
        "available": true,
        "autostart": true,
        "dynamic": false,
        "uptime": 963200,
        "l3_device": "eth0",
        "proto": "dhcpv6",
        "device": "eth0",
        "metric": 0,
        "dns_metric": 0,
        "delegation": true,
        "ipv4-address": [

        ],
        "ipv6-address": [
                {
                        "address": "2a01:7e00::f03c:91ff:fe92:bfb9",
                        "mask": 64,
                        "preferred": 3597,
                        "valid": 14397
                }
        ],
        "ipv6-prefix": [

        ],
        "ipv6-prefix-assignment": [

        ],
        "route": [
                {
                        "target": "::",
                        "mask": 0,
                        "nexthop": "fe80::1",
                        "source": "2a01:7e00:e000:26e::\/64"
                },
                {
                        "target": "2a01:7e00::",
                        "mask": 64,
                        "nexthop": "::",
                        "metric": 256,
                        "valid": 14397,
                        "source": "::\/0"
                },
                {
                        "target": "::",
                        "mask": 0,
                        "nexthop": "fe80::1",
                        "metric": 512,
                        "valid": 297,
                        "source": "2a01:7e00::f03c:91ff:fe92:bfb9\/64"
                }
        ],
        "dns-server": [
                "2001:4860:4860::8888",
                "2001:4860:4860::8844"
        ],
        "dns-search": [

        ],
        "inactive": {
                "ipv4-address": [

                ],
                "ipv6-address": [

                ],
                "route": [

                ],
                "dns-server": [

                ],
                "dns-search": [

                ]
        },
        "data": {

        }
}
```

jow-, 2018-09-18:
Thanks, please provide also the JSON output of http://192.168.1.1/cgi-bin/luci/admin/status/overview?status=1

vgaetera, 2018-09-18:
> {"swap":{"free":0,"total":0},"conncount":61,"wan6":{"mac":"F2:3C:91:92:BF:B9","proto":"dhcpv6","dns":["2001:4860:4860::8888","2001:4860:4860::8844"],"link":"\/cgi-bin\/luci\/admin\/network\/network\/wan6","i18n":"DHCPv6 client","ether":true,"uptime":966692,"ip6addr":"2a01:7e00::f03c:91ff:fe92:bfb9\/64","name":"Ethernet Adapter: \"eth0\"","type":"ethernet","gw6addr":"fe80::1","ifname":"eth0"},"memory":{"buffered":25931776,"total":1043705856,"shared":1183744,"free":962822144},"uptime":966710,"wifinets":[],"wan":{"mac":"F2:3C:91:92:BF:B9","proto":"dhcp","ipaddr":"139.162.227.238","link":"\/cgi-bin\/luci\/admin\/network\/network\/wan","netmask":"255.255.255.0","ether":true,"gwaddr":"139.162.227.1","expires":70098,"uptime":966702,"name":"Ethernet Adapter: \"eth0\"","type":"ethernet","ifname":"eth0","i18n":"DHCP client","dns":["8.8.8.8","8.8.4.4"]},"localtime":"Tue Sep 18 19:45:25 2018","connmax":16384,"loadavg":[0,0,0]}

jow-, 2018-09-20:
Fixed through #2166 / 99265f3f5984bac7f0ba6f2254ee5492581dcaae

closed 2018-09-20 by jow-

vgaetera, 2018-09-20:
@jow- could you apply the fix to 18.06 branch?
It would be nice to see in the upcoming 18.06.2 release.
Thanks.