openwrt/luci #2170: UCI 500 on apply changes, 18.06.1

systemcrash, 2018-09-21:
Supply the following if possible:
 - Device problem occurs on
 - Software versions of OpenWrt/LEDE release, packages, etc.
 - Steps to reproduce
   
device:
(occurs on various devices)

Model	TP-Link Archer C7 v2
Architecture	Qualcomm Atheros QCA9558 ver 1 rev 0
Firmware Version	OpenWrt 18.06.1 r7258-5eb055306f / LuCI openwrt-18.06 branch (git-18.228.31946-f64b152)
Kernel Version		4.9.120

steps to reproduce:
go to Firewall, Traffic Rules, move any rule up or down 1 place, scroll down to bottom, press save. 

click on "unsaved changes" at top:

500 Internal Server Error
Sorry, the server encountered an unexpected error.
=========================================
Failed to execute call dispatcher target for entry '/admin/uci/changes'.
The called action terminated with an exception:
/usr/lib/lua/luci/template.lua:97: Failed to execute template 'admin_uci/changes'.
A runtime error occured: [string "/usr/lib/lua/luci/view/admin_uci/changelog...."]:36: attempt to index local 'o' (a number value)
stack traceback:
	[string "/usr/lib/lua/luci/view/admin_uci/changelog...."]:36: in function 'uci_changelog'
	[string "/usr/lib/lua/luci/view/admin_uci/changes.ht..."]:6: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function 
	(tail call): ?
	/usr/lib/lua/luci/controller/admin/uci.lua:38: in function 
	(tail call): ?

=============

of note: it is possible to run command line uci version still.

yangfl, 2018-10-10:
https://github.com/openwrt/luci/blob/openwrt-18.06/modules/luci-base/luasrc/model/uci.lua#L109 does not correctly `order` command and will insert a number index with no value associated, which causes https://github.com/openwrt/luci/blob/openwrt-18.06/modules/luci-mod-admin-full/luasrc/view/admin_uci/changelog.htm#L53 crash since `o` is a number. But the trunk version of https://github.com/openwrt/luci/blob/master/modules/luci-base/luasrc/view/admin_uci/changelog.htm will not be affected by this (it correctly supports `order` command).

fantuz, 2018-10-30:
So @yangfl is it safe to replace the htm display routine for the trunk version you suggested ?

fantuz, 2018-10-30:
or, a save and apply is still a valid option ?

TippyLion28, 2018-11-18:
So this will be fixed in the next stable release?

jow-, 2021-12-03:
This has been fixed a while ago.

closed 2021-12-03 by jow-