openwrt/luci #2189: First Login, Set Password Button is gone.

froodproton, 2018-09-27:
When I flash my R7800 with Hnymans Build (Script), on the first Luci Login, I get asked to set a password. Normally, there is a Button that leads me to the system --> admin page to enter the new password, but I am missing this button in the master build on 25.09.
master-r8133-61a5994900-20180925

hnyman, 2018-09-27:
Seems that you are right. The button is missing.

[image]

The button code is still in header.htm , sothe button should be visible. My first hunch is that the split of luci-mod-admin-full into three sub-apps might be the reason, as possibly the lookup for "admin/system/admin" fails and button does not get shown. 

https://github.com/openwrt/luci/blob/58d7ee8ab328f0478863bf4aeac64f53a6c9816f/themes/luci-theme-bootstrap/luasrc/view/themes/bootstrap/header.htm#L196

hnyman, 2018-09-27:
@jow- actually fixed it a few minutes ago with https://github.com/openwrt/luci/commit/58d7ee8ab328f0478863bf4aeac64f53a6c9816f

There was a typo.

closed 2018-09-27 by hnyman

cshoredaniel, 2018-09-29:
On 2018-09-27 3:46 p.m., Hannu Nyman wrote:
>
> Seems that you are right. The button is missing.
>
I will try to look into this later this weekend.  I may have missed 
that; IIRC the Material theme had the message but not the button before 
the split, so it may be some other issue.

> image 
> <https://user-images.githubusercontent.com/7926856/46170480-9f6ed280-c2a6-11e8-8361-7c19b0d5e5c6.png>
>
> The button code is still in header.htm , sothe button should be 
> visible. My first hunch is that the split of luci-mod-admin-full into 
> three sub-apps might be the reason, as possibly the lookup for 
> "admin/system/admin" fails and button does not get shown.
>
> https://github.com/openwrt/luci/blob/58d7ee8ab328f0478863bf4aeac64f53a6c9816f/themes/luci-theme-bootstrap/luasrc/view/themes/bootstrap/header.htm#L196
>
If luci-admin-full meta-package is installed (or all three sub-apps) 
there should be no difference from before the split. But will verify 
that there isn't something I missed.

Regards,

Daniel

hnyman, 2018-09-30:
> If luci-admin-full meta-package is installed (or all three sub-apps) there should be no difference from before the split. But will verify that there isn't something I missed.

There was typo (amdin vs. admin) in your commit https://github.com/openwrt/luci/commit/6ec0353201435e0d0d7d32820d8ba600b4ca7b5b#diff-7fa9a24152eff24c800bf0579b65553e 

jow has already fixed that with 58d7ee8 , as I explained above.