openwrt/luci #2203: luci-app-wireguard QRCode Android

markuznw, 2018-10-06:
Hello Dirk and thanks for this nice addition to luci!
I've tried to use it to import data to official Wireguard Android app but it fails with this error, is this on openwrt side or wireguard android side?
Sorry for stupid question but I'm a total newbie with QRCodes :D 
[image]

Thank you @dibdot :)

dibdot, 2018-10-06:
@markuznw 
Hi thanks for the report. 
The bug is on the android side. I've never seen this before - up to now I've only tested with my mullvad account on android 7.1.2 (Lineage os 14.1) and with Oreo 8.1 (Android One). Could you please scan the qr code with a normal qr code reader and post the data output (of course with obfuscated private key!).

@zx2c4 @danrl just for your information

dibdot, 2018-10-06:
@markuznw 
seems to be that you found the root cause - thanks for digging ... same happens if you define an endpoint without endpoint ip address. This is clearly a limitation of the current android client, "(none)" is a valid value and the client should handle that. I will close this LuCI ticket by now.

Thanks again!

closed 2018-10-06 by dibdot