openwrt/luci #2211: Too long zonename in interface firewall settings break firewall

kubco2, 2018-10-09:
In interface firewall configuration new zone is not limited to 11 chars[1] and after saving changes for interface, firewall will stop work if newzone has too long name.

[1] https://wiki.openwrt.org/doc/uci/firewall#zones

closed 2018-10-10 by jow-

kubco2, 2018-10-10:
Hello, I think this wont work. I tried it without data-optional. This only add visual style, do not force to 11chars and allow save and apply.
I tried also fix it in ifaces.lua, but I didnt managed how to propagate nice error message to webif. It only print "something is inccorect" but user do not see red inputbox, like in LEDE17.01, because it is hidden in select box.
This is why I left it without PR.
Another idea was adding maxlength attribute on input box, but this will be not same as on other places.

jow-, 2018-10-10:
It works, I tested it.

kubco2, 2018-10-10:
You are right, thanks. Will test it.

kubco2, 2018-10-10:
It works. Good job.

commit cb7f9d2e (2018-10-10): luci-base: validate name in firewall zone selector widget

When creating a new zone through the firewall zone selector widget,
validate the given name before creating new items.

Depends on commit 0b6ae96f2 ("luci-base: cbi.js: recognize invalid
input in dropdown create field").

Fixes: #2211