openwrt/luci #2212: Switch: changing vid option does not change vlan option

kubco2, 2018-10-09:
Hello,
we are facing this problem on TP-Link tl-wr1043nd v3:
Maximum VLAN number on device is 127. VID can be more than 127.

Problem in webif is if I have already defined VLAN 127 and I want to add new VLAN 81, at first, after clicking "add", it defaults to vid=128, vlan=128, then I am allowed change VID, but this VLAN will never work, because it will fail to create because of invalid VLAN number.

At least allowing change of VLAN in webif should fix it for us. What do you think?

jow-, 2018-10-10:
When a `vid` attribute exists, the `vlan` attribute should be treated as opaque, monotonic increasing identifier. LuCI implements it that way, that it takes the highest existing ID and increments it by one.

You should not declare a VLAN with `vid = 127` and `vlan = 127` in the first place. Simply let it use `vlan = 3` and `vid = 127` to get the desired behaviour.

I will not implement direct control of the `vlan` attribute in LuCI for the reasons stated above.

closed 2018-10-10 by jow-

kubco2, 2018-10-10:
Thanks. I will fix it in our config script.