openwrt/luci #2217: Different country codes in differents radios

HenryGiraldo, 2018-10-10:
OpenWRT sets a global regulatory domain for the full system and takes the last "country" defined in /etc/config/wireless as the regulatory domain.
In LuCI you can set a different country code per radio, but It could not take effect if you have more than one radio and the country code differs.

For example:

```
config wifi-device 'radio1'
	option type 'mac80211'
	option channel '11'
	option hwmode '11ng'
	option path 'platform/qca955x_wmac'
	option htmode 'HT20'
	option disabled '0'
	option country 'DE'

config wifi-device 'radio0'
	option type 'mac80211'
	option channel '40'
	option hwmode '11a'
	option path 'pci0000:00/0000:00:00.0'
	option htmode 'VHT80'
	option disabled '0'
	option country 'HK'
```

with this setup, DE country code will not take effect and the whole system operates in HK country code.

Would it be interesting for the country code to only be set once away from the radios, or when a new one is set on any of the radios, to be applied to all radios?

jow-, 2018-10-10:
The configuration model in OpenWrt should be fixed first. I don't think it makes sense to workaround this in the gui.

yangfl, 2018-10-10:
It's not LuCI which has the problem but UCI itself.

HenryGiraldo, 2018-10-11:
Yes, you're right, but I can't think of any way to solve it in OpenWRT. Maybe creating a unique section in a UCI file for the country code?

Should I report the problem in the OpenWRT project?

yangfl, 2018-10-11:
I don't see a need to address this. One may just want to set different regdoms for different phys, and start different phys according to different situations. As long as phys aren't brought up together, everything would be fine.

jow-, 2018-10-11:
Yes. Imho it would probably make sense to move the radio region settins into /etc/config/system as this is where the other locality settings are stored. But that is just a quick not fully thought through idea.

jow-, 2018-11-21:
Closing this as there won't be any progress on the LuCI side. This is something that needs to be redefined in OpenWrt first.

closed 2018-11-21 by jow-