openwrt/luci #2222: No "Back to Overview" button when using simpleform.htm with form() in controller file

he-ma, 2018-10-11:
When using form() in controller index function instead of cbi() function to avoid the "security token is invalid or already expired" problem (#1722) no "Back to Overview" button is shown anymore.

In my model/cbi file i use
```
local DISP = require("luci.dispatcher")
f = SimpleForm(...)
f.redirect = DISP.build_url(...)
```
By having a look on  simpleform.htm I noticed 
```
...
local display_back   = (redirect)
local display_cancel = (self.cancel ~= false and self.on_cancel)
local display_skip   = (self.flow and self.flow.skip)
local display_submit = (self.submit ~= false)
local display_reset  = (self.reset ~= false)
...
if display_back then
    %><input class="cbi-button cbi-button-link" type="button" value="<%:Back to Overview%>" onclick="location.href='<%=pcdata(redirect)%>'" /> <%
end
...
```
So display_back doesn't refer to **self.redirect** but only redirect.
By changing the redirect to self.redirect it is working again.

Is there any idea how to fix this in the model/cbi or should simpleform.htm be modified?

feckert, 2019-10-11:
We haven't heard back from you, so we're going to go ahead and close the issue.
If you're still experiencing the problem, please re-open the issue.

closed 2019-10-11 by feckert

wolandmaster, 2021-03-05:
I don't know whether it is the intended way to use it, but this is working:
```_G.redirect = luci.dispatcher.build_url("...")```