openwrt/luci #2229: luci-app-statistics does not creat graphs from collectd-mod-snmp

velainur, 2018-10-15:
Hi, I installed luci-app-statistics and collectd-mod-snmp, but seems all rrd files created via collectd, isnt translated via luci-app-statistics.

Please let me know which files do you need in order to clarify the issue

jow-, 2018-10-16:
The `luci-app-statistics` application currently does not support the SNMP plugin.

closed 2018-10-16 by jow-

velainur, 2018-10-16:
Hi,

Thanks for you fast answer. I think luci-app-statistics would be really easy and fast to support collectd-mod-snmp, since i succesfully make luci-app-statistics print "uptime" graph got via a simple symlink and translating default folder structure.

I want to make it possible cloning other folder structure, but i need to understand how luci-app-statistics read folder and sort the information. For example...

i queried via snmp a switch to get packets per interface (in/out). I created several folders "interface-eth0,1,2...n" and inside the folder i created a symlink to the proper rrd file that belongs the "snmp" autocreated folder. I can see the interface structures, but not the graph as i did with the "uptime" rrd file.

could you bring me some ideas?

jow-, 2018-10-16:
You need to provide a data model for SNMP, look into https://github.com/openwrt/luci/tree/master/applications/luci-app-statistics/luasrc/statistics/rrdtool/definitions for examples.

velainur, 2018-10-16:
Amazin, i will try to create this data model and send you back the information so any other could share it.

Anyway, amazin work with this... im really impressed how luci works.

Regards,