openwrt/luci #2231: luci seems to be corrupt, unable to find section 'main'

eduperez, 2018-10-16:
* OpenWrt 18.06.1 on a WRT3200ACM
* Sometimes, when I try to access the main page on the router, I get this error:
```
/usr/lib/lua/luci/dispatcher.lua:230: /etc/config/luci seems to be corrupt, unable to find section 'main'
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:230: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:127: in function </usr/lib/lua/luci/dispatcher.lua:126>

```
* However, "`/etc/config/luci`" looks fine, and just rebooting the device fixes the issue.
* When this happens, all other functionality on the router works fine., but f I log into the router and execute "`logread -l 100`", all I can see is this single line:
```
Thu Jan  1 01:00:00 1970 kern.emerg kernel: 
```
* A couple of minutes later, the logs fill again with _normal_ messages, but luci never recovers.
* Restarting "`uhttpd`" does not fix the issue.

jow-, 2018-10-17:
This can also happen when the `rpcd` service is not running / crashing

eduperez, 2018-10-17:
> This can also happen when the `rpcd` service is not running / crashing

Thanks, that is interesting! I will check the status of `rpcd` when this happens again (unfortunately, I cannot trigger it on demand), and move the bug report somewhere else if this is an issue with that service.

jow-, 2018-10-20:
The symptoms you describe sounds like a system level issue (rpcd crashing / oom etc.) not something LuCI specific. Will close this for now.

closed 2018-10-20 by jow-

eduperez, 2018-10-23:
Can confirm this looks like an OOM issue: after adding some swap to the device, I could not reproduce it; however, it happens immediately when I remove the swap.

Thanks!

msangel, 2018-12-27:
>after adding some swap to the device

any link / instruction how exactly you fixed that?

eduperez, 2018-12-28:
Any guide about adding swap to OpenWrt will do, really: https://openwrt.org/docs/guide-user/storage/fstab?s[]=swap

token0, 2019-04-28:
`service rpcd restart`
works for me

JocelynDelalande, 2020-04-10:
I had the same error, after an upgrade from some LEDE version (from 2017 IIRC) to OpenWRT 18.06.08. I choosen to keep settings while upgrading, that was apparently a mistake.

To fix it, I did :

- run by hand (SSH console) : `rcpd` (`/etc/init.d/rcpd`  did not work anymore)
- restore factory defaults

manokara, 2020-07-25:
No need to restore factory defaults (although it might be a good idea, specially if you don't have that many settings). You can add `rpcd &` to `/etc/rc.local` before the `exit 0` line and  everytime you boot `rpcd` will start and LuCi will work just fine.

pezoli, 2020-08-08:
Another way to solve:
https://forum.openwrt.org/t/luci-webgui-crash-after-updating-to-ver-18-6-8-on-tp-link-tl-wa801n-v2/57620/9

wanghongxun226, 2022-02-18:
> `service rpcd restart` works for me

me too

8666, 2022-07-13:
at some version 
/var/run/ubus.sock
was moved to
/var/run/ubus/ubus.sock

so edit /etc/config/rpcd

cantalupo555, 2022-12-24:
> `service rpcd restart` works for me

Thanks!

ulope, 2024-01-14:
I just had this happen as well but it can't be OOM since this router has 4GB RAM with only ~200MB used.

So this sounds like some sort of bug...

FeroxTL, 2025-09-25:
@fahadshery try this:
```sh
ssh -oHostKeyAlgorithms=+ssh-rsa root@192.168.1.1
```