openwrt/luci #2232: OpenWRT 18.06 and wireguard

erotavlas85, 2018-10-19:
Hi,
I noticed a strange behaviour of wireguard 0.0.20180718-2 with openWRT OpenWrt 18.06.1 r7258-5eb055306f.
```
opkg install wireguard kmod-wireguard luci-app-wireguard luci-proto-wireguard wireguard-tools
Package wireguard (0.0.20180718-2) installed in root is up to date.
Package kmod-wireguard (4.9.120+0.0.20180718-2) installed in root is up to date.
Package luci-app-wireguard (git-18.284.42397-55ebe88-1) installed in root is up to date.
Package luci-proto-wireguard (git-18.284.42397-55ebe88-1) installed in root is up to date.
Package wireguard-tools (0.0.20180718-2) installed in root is up to date.
```

The problem is that if I use only one peer, all works well. While if I add more peers the LUCI configuration under interface->wireguard interface name is different from the one of wireguard.

```
root@BTHomeHub:/# wg showconf wireguard
[Interface]
ListenPort = 51820
PrivateKey =

[Peer]
PublicKey = fEq76eJ+YYvOZMZB+5TVJ0T1mlRqdTKl1kWZsOQq111=
AllowedIPs = 0.0.0.0/0, ::/0
Endpoint = IP:3032
PersistentKeepalive = 25

[Peer]
PublicKey = SzmAUqFzWRHazql4EGVgphTdUBFq8Wus980nCl99111=
## missing line
Endpoint = IP:51820
PersistentKeepalive = 25

## missing peer
```

I did not find any file under` /etc/config `as usual.

```
wg
interface: wireguard
  public key: xhrX4aXDNUw2eCygHnXQYuaYpRKemY1HbshLATzI111=
  private key: (hidden)
  listening port: 51820

peer: fEq76eJ+YYvOZMZB+5TVJ0T1mlRqdTKl1kWZsOQq111=
  endpoint: IP:3032
  allowed ips: 0.0.0.0/0, ::/0
  transfer: 0 B received, 46.11 KiB sent
  persistent keepalive: every 25 seconds

peer: SzmAUqFzWRHazql4EGVgphTdUBFq8Wus980nCl99111=
  endpoint: IP:51820
  allowed ips: (none) ## missing
  transfer: 0 B received, 38.73 KiB sent
  persistent keepalive: every 25 seconds

## missing a peer
```
Best Regards

P.S.
Also the field preshared key seems not work properly with more than one peer.

jow-, 2018-10-20:
@danrl - any idea?

danrl, 2018-10-20:
I am confused that nothing under `/etc/config` shows up. The file responsible for writing the actual wireguard config based on the files that should be under `/etc/config` has not really changed much. https://github.com/openwrt/openwrt/commits/34e22653ac18b6ac7fd368ca47625f665808067f/package/network/services/wireguard/files/wireguard.sh

Is the problem reproducible even if you reboot every time after adding a peer? I am trying to isolate the problem. It could be how the UCI config it (not) written or it could be how the wg.conf is written based on what is (or is not) in UCI. This could tell us if we have to look at the `wireguard` package or the `luci-proto-wireguard` package.

erotavlas85, 2018-10-21:
I confirm the issue after several trials and reboot.
```
ls /etc/config/
adblock              luci                 sqm
adblock-opkg         luci_statistics      sqm-opkg
ddns                 minidlna             system
ddns-opkg            minidlna-opkg        transmission
dhcp                 mwan3                transmission-opkg
dhcp-opkg            mwan3-opkg           tvheadend
dnscrypt-proxy       network              ucitrack
dnscrypt-proxy-opkg  nlbwmon              udpxy
dropbear             openvpn              uhttpd
firewall             openvpn-opkg         uhttpd-opkg
firewall.save        openvpn.bak          vnstat
hd-idle              openvpn_recipes      vnstat-opkg
hd-idle-opkg         qos                  wireless
igmpproxy            rpcd                 znc
```

danrl, 2018-10-22:
Ok, so just to be sure it is not the wireguard package that has the bug: If you create your desired multi-peer configuration in UCI, it does get translated correctly and shows up correctly in `wg show`, right?

To test that, it may be helpful to remove luci-proto-wireguard during testing.

erotavlas85, 2018-10-22:
No, it does not show up correctly via `wg show`. I also tried to remove `luci-proto-wireguard`, but I got this:
opkg remove luci-proto-wireguard
No packages removed.
Collected errors:
 * print_dependents_warning: Package luci-proto-wireguard is depended upon by packages:
 * print_dependents_warning: 	luci-app-wireguard
 * print_dependents_warning: These might cease to work if package luci-proto-wireguard is removed.

 * print_dependents_warning: Force removal of this package with --force-depends.
 * print_dependents_warning: Force removal of this package and its dependents
 * print_dependents_warning: with --force-removal-of-dependent-packages.

KuriKai, 2018-10-30:
I seem to be having the same issue that the allowed ip's are set to (none)
1st peer was fine for wgguest till i added the 2nd peer for wgguest
2nd peer was fine till i added the 3rd peer for wgguest
3rd peer is working

my private wg0 is working fine with only one peer, even though the wgguest has peers

```
kmod-wireguard 4.14.54+0.0.20180718-2
luci-app-wireguard git-18.219.52706-a49e4e4-1
luci-proto-wireguard git-18.219.52706-a49e4e4-1
wireguard-tools 0.0.20180718-2
```

the wireguard config is in the /etc/config/network file

```
root@OpenWrt:/etc/config# cat network 

...

config wireguard_wg0
	option route_allowed_ips '1'
	option endpoint_port '51820'
	option persistent_keepalive '25'
	option endpoint_host '192.168.1.1'
	option public_key '###'
	list allowed_ips '0.0.0.0/0'
	list allowed_ips '::0/0'

config interface 'wg0'
	option proto 'wireguard'
	option private_key '###'
	option listen_port '51820'
	list addresses '10.0.1.1/32'

config interface 'wgguest'
	option proto 'wireguard'
	option private_key '###'
	option listen_port '51821'
	list addresses '10.0.2.1/32'

config wireguard_wgguest
	option public_key '###'
	option route_allowed_ips '1'
	option persistent_keepalive '25'
	option description 'mobile_10.0.2.2'
	list allowed_ips '0.0.0.0/0'

...

config wireguard_wgguest
	option public_key '###'
	option persistent_keepalive '25'
	option route_allowed_ips '1'
	option description 'mobile_10.0.2.3'
	list allowed_ips '0.0.0.0/0'

config wireguard_wgguest
	list allowed_ips '0.0.0.0/0'
	option route_allowed_ips '1'
	option persistent_keepalive '25'
	option public_key '###'
	option description 'macbook_10.0.2.4'
```

Here is the "wg show"

```
interface: wgguest
  public key: ####
  private key: (hidden)
  listening port: 51821

peer: ###Ds=
  endpoint: 60.216.105.171:8606
  allowed ips: 0.0.0.0/0
  latest handshake: 37 seconds ago
  transfer: 5.27 MiB received, 26.87 MiB sent
  persistent keepalive: every 25 seconds

peer: ###3W0=
  endpoint: [###]:35714
  allowed ips: (none)
  latest handshake: 9 minutes, 31 seconds ago
  transfer: 952 B received, 22.70 KiB sent
  persistent keepalive: every 25 seconds

peer: ###ys=
  allowed ips: (none)
  persistent keepalive: every 25 seconds

interface: wg0
  public key: ####
  private key: (hidden)
  listening port: 51820

peer: ###4So=
  endpoint: [###]:55456
  allowed ips: 0.0.0.0/0
  latest handshake: 5 minutes, 20 seconds ago
  transfer: 13.39 KiB received, 62.42 KiB sent
  persistent keepalive: every 25 seconds

```

danrl, 2018-10-30:
Thanks for the bug report! Did you use luci? Would you have the same issue
with plain UCI?

KuriKai, 2018-10-31:
I used luci to configure it.

I'll have to test without luci. Do you happen to know how to set wireguard up and down without luci?

erotavlas85, 2018-11-04:
I removed both `luci-proto-wireguard` and` luci-proto-wireguard`. I manually modified the configuration file `/etc/config/network`:

```
config interface 'Wireguard'
	option proto 'wireguard'
	option private_key 'sDHHB/Wk0dGpdev9TSxpHgRBrTNPzsyjFb5Gf3AG***'
	list addresses '10.92.13.19/32'
	list addresses 'fc00:bbbb:baab:bb01::d13/128'
	option force_link '1'
	option listen_port '51820'

config wireguard_Wireguard
	option public_key 'fEq76eJ+YYvOZMZB+5TVJ0T1mlRqdTKl1kWZsOQq***'
	list allowed_ips '0.0.0.0/0'
	list allowed_ips '::0/0'
	option route_allowed_ips '1'
	option endpoint_host '93.114.64.42'
	option endpoint_port '51820'
	option persistent_keepalive '25'

config wireguard_Wireguard
	option public_key 'fEq76eJ+YYvOZMZB+5TVJ0T1mlRqdTKl1kWZsOQq***'
	list allowed_ips '0.0.0.0/0'
	list allowed_ips '::0/0'
	option route_allowed_ips '1'
	option endpoint_host '185.112.170.226'
	option endpoint_port '3032'
	option persistent_keepalive '25'
```
Unfortunately, the command `wg` or `wg show` only one interface. In particular, the last one and the device connects to it.
It seems a wireguard bug...

danrl, 2018-11-10:
Thanks for all the reports, everyone! I finally had time to rebuild my test environment.
Testing in LuCI openwrt-18.06 branch (git-18.228.31946-f64b152) / OpenWrt 18.06.1 r7258-5eb055306f in Virtual Box (x86) using the 64bit generic combined image.

Good (or bad?) news? I was not able to reproduce the problems.

Here are my steps:
- add interface "foo" type "Wireguard VPN"
- add peer 1
- add peer 2
- add peer 3
- add interface "bar" type "Wireguard VPN"
- add peer 1

After each step I **saved the config** and **restarted the interface in question**. The config file was written as expected in `/etc/config/network` and `wg` showed the configuration as expected.

Here are my resulting configs for future reference. Keys are obviously randomized :) 

```
config interface 'loopback'
	option ifname 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fd1d:a5f0:a31a::/48'

config interface 'lan'
	option type 'bridge'
	option ifname 'eth0'
	option proto 'static'
	option ipaddr '192.168.1.1'
	option netmask '255.255.255.0'
	option ip6assign '60'

config interface 'wan'
	option proto 'dhcp'
	option ifname 'eth1'
	option auto '0'

config interface 'foo'
	option proto 'wireguard'
	option private_key 'sE/W8u/A0cHQyPK7hFAASsSPxatWUNlooRPQfD92KW8='
	list addresses '100.64.0.1/24'

config wireguard_foo
	option public_key 'aG7QGqHv4j45WNqZsouWrGx8gn4J8PCD2x5Q0HqsdCI='
	list allowed_ips '100.64.1.0/24'
	option route_allowed_ips '1'

config wireguard_foo
	option public_key 'VjUlHw/GHjbThDb3hNir/wUOaM9S1apOanB+osD63Xk='
	list allowed_ips '100.64.2.0/24'

config wireguard_foo
	option public_key 'VhmD5nql3bXbmoPiwWjpjqm7XOLRKO0+NIMJDKJHkRw='

config interface 'bar'
	option proto 'wireguard'
	option private_key 'YKs4oX0iKoZ7q2zCOShXJ2MPJf3p2RqMVupncm5hA20='

config wireguard_bar
	option public_key 'aYNhCwLmUw7NncLZH/plJN9Qp3MQv1SuX8FPk1fTmy4='

```

```
interface: foo
  public key: 90SueDKkIP6Vqy2iMOnEakZe+RuOurfMxMfmQJQDTHU=
  private key: (hidden)
  listening port: 37506

peer: aG7QGqHv4j45WNqZsouWrGx8gn4J8PCD2x5Q0HqsdCI=
  allowed ips: 100.64.1.0/24

peer: VjUlHw/GHjbThDb3hNir/wUOaM9S1apOanB+osD63Xk=
  allowed ips: 100.64.2.0/24

peer: VhmD5nql3bXbmoPiwWjpjqm7XOLRKO0+NIMJDKJHkRw=
  allowed ips: (none)

interface: bar
  public key: JB/HmzBcDHUSaPvsoHlxZ2aHcy2hfo7da7vOv2efF3s=
  private key: (hidden)
  listening port: 39773

peer: aYNhCwLmUw7NncLZH/plJN9Qp3MQv1SuX8FPk1fTmy4=
  allowed ips: (none)
```

I'll leave this open for further reports, just in case we find a way to reproduce this in the future.

erotavlas85, 2018-11-10:
@danrl Thank you for testing. The only difference that I see is on the different platform x86 vs mips form my router. It could be a problem?

danrl, 2018-11-10:
@erotavlas85 I don't really think the architecture makes a difference here. That being said, there is always a little surprise hidden somewhere that can be tracked down to unexpected architecture behavior.

erotavlas85, 2018-11-11:
@danrl of course, but I do not see any other reason. Since I also tried with a fresh install of the same openWRT version on my device. Maybe different platform have different installed packages or different version? I will wait for 18.6.2 release and I will try again.

KuriKai, 2018-11-20:
@danrl does it still work for you if you are using the ip range 0.0.0.0/0 for all 3 of the peers?

danrl, 2018-11-20:
@KuriKai I have not tried that. Does that even make sense? How do you route back traffic, if you have three default routes? How do you know which peer shall receive the traffic that goes back? I am confused. Could you elaborate on the use case and what the expected result of multiple peers with a netmask of /0 would be?

KuriKai, 2018-11-20:
According to https://www.wireguard.com/ Having "0.0.0.0/0" in the allowed ip's is used as a wildcard, meaning it will accept a connection from anywhere (useful for connecting mobile devices to the wireguard server).
https://wiki.archlinux.org/index.php/WireGuard#Peer_A_setup also has some information about 0.0.0.0/0 allowing any ip to connect

KuriKai, 2018-11-20:
From the wireguard webpage
[image]
From the arch wiki
[image]

danrl, 2018-11-20:
I think you might be confusing the inner and the outer header's source address. Allowed-IPs applies to the inner header, which is read and validated after decrypting.

I created a graphic to explain this once. Just scroll down to the figure showing IP headers and encapsulation here: https://danrl.com/blog/2016/travel-wifi/#vpn

KuriKai, 2018-11-21:
ok, I changed on the vpn server(openwrt) under the peers, to have allowed ip of the clients address(10.0.2.2/32|10.0.2.3/32) gave openwrt a reboot, and it works.

danrl, 2018-11-21:
@jow- I think we can close this.

erotavlas85, 2018-11-21:
@danrl in my case with a fresh install of the openWRT on my router, the problem is still present.

KuriKai, 2018-11-21:
@erotavlas85 
How many of the AllowedIPs for your peers are set to "0.0.0.0/0, ::/0"?
I had my three peers all set to:
peer1
allowed ips "0.0.0.0/0"
peer2
allowed ips "0.0.0.0/0"
peer3
allowed ips "0.0.0.0/0"

and that was causing the issue for me.
I changed them to the ip of the address of the client:
peer1
allowed ips 10.0.2.2/32
peer2
allowed ips 10.0.2.3/32
peer3
allowed ips 10.0.2.4/32

and the problem was solved

erotavlas85, 2018-11-21:
@KuriKai 
I'm using openWRT as client for VPN connection provided by different providers (azireVPN and mullvad).
They suggested to use AllowedIPs = 0.0.0.0/0,::0/0.
The same configuration works well both on GNU/linux and android.
If I only use one peer, it works well.

metronidazole, 2018-11-23:
You need to restart the wireguard interface after making changes in LuCi.

erdoukki, 2019-03-07:
multiple peers cannot work with the actual internal naming convention ...
from the official documentation it is said;  

> The name of a peer section must be wireguard_XX where XX is the name of the wireguard interface section. 
Ref : https://openwrt.org/docs/guide-user/network/tunneling_interface_protocols?s[]=wireguard#protocol_wireguard_wireguard_vpn

and in my config, when I add peers, they get the same name... reflecting the up said information...

May be it is the actual problem reported here ?

hope it will help to fix this issue...

erdoukki, 2019-04-05:
> multiple peers cannot work with the actual internal naming convention ...
> from the official documentation it is said;
>
>
> and in my config, when I add peers, they get the same name... reflecting the [...]

My fault and misanderstood of the WireGuard !
I get all working after some help at the OpenWRT forum and with few tweaks.
Sorry for the noise !!!

closed 2019-04-05 by jow-

erotavlas85, 2019-04-06:
I have to try with the latest OpenWRT 18.06.2 in order to confirm that the problem is solved. At the moment, I have not time for this.