openwrt/luci #2255: luci-app-olsr: updates to new version of jsoninfo plugin

pmelange, 2018-11-04:
The OLSR jsoninfo plugin has a new version.  The many things have changed which make the luci-app-olsr status pages no longer work properly.  For example ```echo /status | nc 127.0.0.1 9090``` no longer works as /status is no longer available.  Also, the field names have changed in almost all request types.

I have spent a few hours trying to figure out how luci and everything works.  After all my efforts, I have gotten nowhere.  

@jow-, could you help us solve this issue?  It would be great to have the next freifunk-berlin firmware finished and this is one of the things blocking us.  If you can/will help, let me know how I can support you.  For example, I could set up a couple routers with the old and new versions of the jsoninfo plugin to help with development.

Here is the issue at freifunk-berlin: https://github.com/freifunk-berlin/firmware/issues/612

(As a side note, the columns of the table headers and the data below do not align any more).

pmelange, 2018-11-05:
I have some preliminary commits at https://github.com/pmelange/luci/tree/luci-app-olsr-new-json-lib

I managed to get all the status pages working again.  I also added some functionality to network.lua in luci-base.  I have not modified the Makefile yet.  I would like to have an opinion on the version number.

Also, I still don't have the table header and columns lining up.  I could really use some help here.

jow-, 2018-11-05:
@pmelange - if you arrange for access I can take a look during the bext few days.

pmelange, 2018-11-07:
@jow, I have sent you a private message about setting up access.  I assume you want to see a router with these changes in a functioning environment.

I have also made PR #2259 with these changes.

pmelange, 2018-11-14:
merged in commit https://github.com/openwrt/luci/commit/ceb342dc8de0235686bfceae7cd3948cb3c2f527

closed 2018-11-14 by pmelange