openwrt/luci #2261: Regression: can't add/customize value entries any longer

dibdot, 2018-11-09:
Hi @jow,

I haven't tracked this down (maybe changes in cbi.js?!), but with current snapshots I can't customize existing values any longer, e.g. a click on "-- custom --" don't load the value for further customization.
This seems to be independent from browsers or themes.

[image]

jow-, 2018-11-09:
Did you try to enter a new value and press enter?

dibdot, 2018-11-09:
No, in the past I could select/pre-load e.g. the first entry and than click on "-- custom --" to edit/further customize the pre-loaded values and that doesn't work now.

dibdot, 2018-11-09:
OK, I'll try to change the app behaviour accordingly - thanks for clarification.

closed 2018-11-09 by dibdot

jow-, 2018-11-09:
Ok, sorry for the trouble. Let me know when I can help