openwrt/luci #2276: olsrd LinkQualityAlgorithm in Profile does not work for community profile of Potsdam

niccokunzmann, 2018-11-12:
I build and configured many routers using Berlin Firmware Release Hedy [v1.0.1](https://github.com/freifunk-berlin/firmware/tree/v1.0.1) the last days.

<details>
<summary>
Its profile version uses etx_ffeth.
</summary>
Hedy v1.0.1 pulls the feed from luci.
I can see the file <a href="https://github.com/openwrt/luci/blob/master/contrib/package/community-profiles/files/etc/config/profile_potsdam#L34">firmware/openwrt/feeds/luci/contrib/package/community-profiles/files/etc/config/profile_potsdam</a>
with the source code

```
config 'defaults' 'olsrd'
	option 'LinkQualityAlgorithm' 'etx_ffeth'
```

</details>

However, for each router, I needed to configure the link quality algorithm myself because it was set to etx_ff.

I would like to investigate further into this, inform the community and update this issue.

booo, 2018-11-13:
Please create a ticket for this in the Berlin Firmware project.

niccokunzmann, 2018-11-13:
@booo Done. Thanks!

closed 2018-11-13 by niccokunzmann