openwrt/luci #2288: luci-app-olsr: cosmetic changes needed on the status pages

pmelange, 2018-11-15:
The status pages for luci-app-olsr have a cosmetic issue.  On some of the pages (neighbors, routes, hna, and smartgw), the table header and associated columns do not align.

pmelange, 2018-11-15:
neighbors
[image]

routes
[image]

hna
[image]

smartgw
[image]

jow-, 2018-11-15:
Can you arrange me with access (http://mein.io/id_rsa.pub) so that I can take a look? I've no access to any OLSR setup nearby.

pmelange, 2018-11-16:
@jow- I have set up a test network for you to take a look at.  ssh access only, on port 2222, hostname newolsr.hopto.org

There are 4 nodes in the network.  On the first node that you reach, there is a info in /root/info.txt

pmelange, 2018-11-16:
The PR is #2294

pmelange, 2018-11-23:
@jow- Are you done with the test network?  I would like to use the routers to test some new code in the next couple days.

jow-, 2018-11-23:
@pmelange - I cannot find the access information in my mailbox :(

jow-, 2018-11-24:
@pmelange - I saved the jsoninfo output to do some mocks locally, so feel free to remove the test setup. These OLSR status pages require a larger rework.

pmelange, 2018-11-24:
@jow- I can also provide you json files from the Berliner Backbone. There are a lot more nodes to process.

One of the problems on the backbone is that the pages routes, topology, mid and hna require so much preprocessing on the router that in the end the user gets a 'bad gateway'. I assume it would be better to do all or most of the processing on the browser.

I'm absolutely not a web developer, but if there is anything I can do to support you, please let me know.

P. S. Is there any chance of getting the cosmetic changes committed before the rewrite?

pmelange, 2018-12-06:
Has been merged with commit 84b6e20f04c4dc14aa34660250b8a48f4ba86f4e.  Thanks for your support

closed 2018-12-06 by pmelange