openwrt/luci #2303: Custom DHCP hostnames only shown for IPv4

bmos, 2018-11-17:
I'm running **OpenWrt SNAPSHOT r8463-59ff8687c7** on a **GL-iNet B1300** and noticed that **Active DHCPv6 Leases** doesn't use the hostnames I've defined in **Network >> DCHP and DNS** but **Active DHCP Leases** does. This is obviously inconsistent.

bmos, 2018-11-17:
Just noticed that some devices do use the names from the **Static Leases** table but not all of them...

ryu0, 2018-11-19:
Are you sure your IPv6 hosts are using DHCPv6? Most of my hosts use SLAAC, so they don't show up under DHCPv6 leases.

bmos, 2018-11-19:
Thanks, that explains why the DHCPv6 list is so short; I was wondering about that.
It doesn't address the key issue I'm mentioning, which is present in the clients which do show up in the list. Some of them (probably the ones that aren't reporting a hostname) show up with the reported hostname rather than the one defined in the DHCP and DNS page in LuCI. On its own, this really an issue but the priority of these two available names is handled differently in the DHCP V4 section.

wblew, 2019-02-01:
I was looking at the situation on my wrt3200acm. It includes both static and dynamic IPv6 leases via odhcpd.

a) I describe the "static IPv6 leases" as those whose "IPv6-Suffix (hex)" has been given a number. Those leases' host names appear in the "Active DHCPv6 Leases". I am describing those whose "IPv6-Suffix (hex)" is blank as dynamic leases.

Static leases have host names in the Active DHCPv6 Leases list, while the dynamic leases their host names.

b) The odhcpd lease file is: /tmp/hosts/odhcpd.leases. While that file contains comment lines for all active IPv6 leases. It only contains an actual 'lease line' for those whose "IPv6-Suffix (hex)" field was defined.

Perhaps there is a correlation between a) and b) above? Perhaps this issue is related to odhcpd's implementation and how its lease file is being interpreted by luci?

closed 2021-10-03 by bmos