openwrt/luci #2305: luci-app-openvpn: FileUpload fields not shown

mwarning, 2018-11-18:
In the OpenVPN LuCI interface, an instance created from a template has many FileUpload fields that are not shown despite being present in the uci config object (e.g. `dh` or `ca`).
They can be added explicitly so that they appear. But removal does not seem to be possible.

dibdot, 2018-11-19:
@jow- I could fix that with the "unsafeupload" property but it looks like some css magic is needed (at least in bootstrap theme):
OpenWrt theme:
[image]

Bootstrap theme:
[image]

Thanks!

mwarning, 2018-11-19:
Would that help? I think this fixes some things.
```
--- a/applications/luci-app-openvpn/luasrc/model/cbi/openvpn-advanced.lua
+++ b/applications/luci-app-openvpn/luasrc/model/cbi/openvpn-advanced.lua
@@ -247,6 +247,10 @@ for _, option in ipairs(params) do
 
 		o.optional = true
 
+		if option[1] == FileUpload then
+			o.datatype = "file"
+		end
```

dibdot, 2018-11-20:
@mwarning No, that will already checked by the used FileUpload function. Appropriate changes for your issue are in #2307

mwarning, 2018-11-21:
As a heads up, I applied the changes from #2307 to the current nightly build of OpenWrt and it did not seem to work yet.

dibdot, 2018-11-21:
@mwarning maybe you could elaborate what "did not seem to work" mean? 
#2307 add the opportunity to remove existing entries ... it doesn't change the (desired) FileUpload behaviour that only valid file handles are shown.

dibdot, 2018-11-21:
That's a feature, not a bug. The FileUpload types are only visible if they reference a valid file (typically uploaded by the user in /etc/luci-uploads/....)

mwarning, 2018-11-21:
I tried to set `secret` and thought that would work. But OpenVPN did not start because the `ca` field etc. contains a dummy file path. But since I do not see that field, I do not know that I have to add (to make it visible) and then remove that field.

mwarning, 2018-11-21:
@dibdot looks good. I can see all fields that are set now. When I clear the `dh` filed and hit "Save & Apply", the field is removed. (But hitting the "Save" button doe not remove an empty field - is that intentional?)

dibdot, 2018-11-21:
Yes, that's the usual behaviour - only "apply" will finally update/commit the changes to the uci file itself.

mwarning, 2018-11-21:
Alright. And is it intentional that luci-app-openvpn does not depend on package openvpn?

dibdot, 2018-11-21:
yep - I will merge #2307 now. Thanks for your input & testing efforts.

mwarning, 2018-11-21:
ok, thanks for your efforts.

closed 2018-11-21 by dibdot