openwrt/luci #2332: Freifunk Wizard: password revealed when not using mouse to go to "next" page

pmelange, 2018-11-26:
With a fresh install of the Freifunk firmware, the wizard comes up and promts the user at the first step to enter a password.  

Without using the mouse, entering the password in the first field, hitting Tab (twice since the UI tweaks), then entering the password again and then hitting Enter causes the password in the first field to be revealed.  The password is the second field remains hidden.

When using a mouse, this does not happen.

jow-, 2018-11-27:
Which LuCI package do I need to install to find this wizard?

pmelange, 2018-11-28:
The package luci-app-ffwizard-berlin resides in https://github.com/freifunk-berlin/firmware-packages

The changePassword code is https://github.com/freifunk-berlin/firmware-packages/blob/master/utils/luci-app-ffwizard-berlin/luasrc/model/cbi/freifunk/assistent/changePassword.lua

It may be easier to just use one of our development images.  Here are links to a few different architectures
https://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/1003/default
https://buildbot.berlin.freifunk.net/buildbot/unstable/x86-generic/926/default
https://buildbot.berlin.freifunk.net/buildbot/unstable/ramips-mt7621/916/default

If you want to test with a different arch, just let me know which one.

feckert, 2019-09-19:
This an issue from this LuCI package repository. Please open an issue on their project page.
https://github.com/freifunk-berlin/firmware-packages

closed 2019-09-19 by feckert