openwrt/luci #2340: Creating a new Port Forward automatically changes protocol from UDP to TCP

hailfinger, 2018-11-27:
Environment: LuCI openwrt-18.06 branch (git-18.329.26825-5e87083)

How to reproduce:
Network -> Firewall -> Port Forwards -> New Port Forward
Enter name "foo"
Enter Protocol "UDP"
Enter External zone "wan"
Enter External port "443"
Click anywhere in the GUI (somewhere in the white space, or even in the field Internal port).
BUG! Protocol automatically changes to "TCP"
Go back and change Protocol to "UDP".
Click on External port form field.
Click anywhere in the GUI
BUG! Protocol changes to "TCP" again.

Apparently leaving the External port form field triggers a changing the protocol to "TCP".

dibdot, 2018-11-28:
It's not a bug, it's a feature ... it's meant as a hint if you enter a standardized tcp port ... just switch it back to udp and click on save & apply. For reference the respective code snippet:

```
			var hints = {
			/*  port    name     0=both, 1=tcp, 2=udp, 3=other */
				21:   [ 'FTP',   1 ],
				22:   [ 'SSH',   1 ],
				53:   [ 'DNS',   0 ],
				80:   [ 'HTTP',  1 ],
				443:  [ 'HTTPS', 1 ],
				3389: [ 'RDP',   1 ],
				5900: [ 'VNC',   1 ],
```

closed 2018-11-28 by dibdot

jow-, 2018-11-28:
It actually is kind of unexpected. I think we should simply remove these hinting code.

hailfinger, 2018-11-28:
Indeed it is unexpected. I wanted to redirect QUIC traffic (UDP/443).

dibdot, 2018-11-28:
Here we go ... ;-)