openwrt/luci #2349: Uploaded snapshot firmware footer corruption

shunjou, 2018-12-02:
I have recently run into the issue where any snapshot firmware image uploaded through luci's flashops page results in the corruption of the image, verified by mismatched checksums. It seems to be stripping the signature from the footer of the image. I confirmed this on multiple devices (ath79 and ipq806x). Curiously, the same does not happen when uploading older 18.06.1 release builds.

Test case using recent official snapshot build: https://drive.google.com/file/d/1H7cwt5D4crHzhBa4UD9L3qvbV_1wYCJV

jow-, 2018-12-02:
Which version are you running? I can't reproduce it on any of my devices here.

shunjou, 2018-12-02:
Master (git-18.333.34665-4f2b80e) on both devices.

jow-, 2018-12-02:
Which version of OpenWrt?

shunjou, 2018-12-02:
OpenWrt SNAPSHOT r8614-78ca6a5 on both devices.

jow-, 2018-12-02:
Cannot verify it on either a Netgear R6100 nor an x86 appliance, unfortunately I have no idea whats wrong there.

hnyman, 2018-12-02:
Sounds similar that I have noticed (and sent email yesterday):

I suspect some misbehaving when using the current LuCI for sysupgrade. Latest trying to sysupgrade from r8614-78ca6a5578.

I have noticed now twice that the transferred sysupgrade image file /tmp/firmware.img is a few bytes short and the sha256sum does not match the expected value.

I have tried with several browsers (Firefox and Edge), and similar results.

Looking at the image file in the buildhost, Windows PC and router, it looks like that the image file transfer from PC via browser skips the last bytes and cuts the image in /tmp to even 0x843000 bytes (8663040 as decimal). Correct size would be 8663230 dec = 0x8430be bytes.

I suspect that this might be due to the recent uhttpd changes, but have no actual evidence to support that.

shunjou, 2018-12-02:
Ah sorry, I neglected to mention that it's my own build. Could my package configuration have any affect?

jow-, 2018-12-02:
None of the uhttpd changes look even remotely related to the problem.

hnyman, 2018-12-02:
Yeah, I know. This has surfaced in the last few days.
Curiously, my devices are ipq806x and ar71xx based.  (havenät tested on my mvebu router, yet)

hnyman, 2018-12-02:
I see the error when trying to sysupgrade R7800 (ipq806x) from:
master-r8580-c6aa4f53e8 of 20181127
master-r8614-78ca6a5578 of 20181130

But sysupgrading from 18.06 or from an older master build master-r8466-251c350727 of 20181116 works ok.

I do not have the intermediate builds in storage any more, but I did several builds in between and sysupgrading R7800 worked ok.

By looking at the image file in /tmp, it looks like the metadata (that is appended to the image in ipq806x sysupgrade) gets stripped away from the end of the image in the checksum verification step (?), or something like that.

And that makes me to suspect this commit: 
https://github.com/openwrt/openwrt/commit/41770add03ad77a0ce41ed424ad050238f7d9272
 " base-files: fwtool: Fix wrong checksum on combined-image with metadata "

@blogic has accepted and merged it on Nov 26th , so it falls into the possible range of culprit commits.

That commit removes the appended metadata from the image. Apparently that mixes ipq806x sysupgrade in LuCI.

hnyman, 2018-12-02:
Manually editing /lib/upgrade/fwtool.sh and reverting that addition of "-t" option makes the firmware.img ok again. Metadata is there and the checksum shown in LuCI matches expectations.

Looks like that https://github.com/openwrt/openwrt/commit/41770add03ad77a0ce41ed424ad050238f7d9272 should be reverted, as the change breaks ipq806x sysupgrade in LuCI

```
root@router1:~# hexdump -C /tmp/firmware.img  | tail -n 20
00841a70  21 e9 a6 f4 c4 4a d7 8e  69 8d b5 0b c9 2d a9 b8  |!....J..i....-..|
00841a80  46 13 b7 d4 5f 08 19 43  83 c4 9f df 33 be 4d 5b  |F..._..C....3.M[|
00841a90  5e 2c 32 68 13 d5 01 09  2e 2f 97 ed 41 3f 84 21  |^,2h...../..A?.!|
00841aa0  58 9c e1 37 42 12 ca a3  fa 08 68 f5 02 7d e9 3f  |X..7B.....h..}.?|
00841ab0  1f 83 78 d9 87 89 43 bb  98 dc ca 0a e7 27 6c 65  |..x...C......'le|
00841ac0  de d1 45 e9 55 a2 60 0d  24 66 45 b1 57 53 96 6e  |..E.U.`.$fE.WS.n|
00841ad0  fc d0 c5 6a 35 64 6d ea  60 f3 74 a3 fe 14 cd 46  |...j5dm.`.t....F|
00841ae0  b4 d5 09 fc ad 47 e4 32  7b 51 41 ec 7f a6 63 cd  |.....G.2{QA...c.|
00841af0  64 33 c8 b9 7e 8e 09 ed  ca 27 36 37 da df 43 cc  |d3..~....'67..C.|
00841b00  f2 a2 bd f8 0a 0b b9 ae  19 44 0d 1e 58 38 79 19  |.........D..X8y.|
00841b10  09 90 78 86 41 26 73 1d  f5 3d 10 dd 0b 28 07 43  |..x.A&s..=...(.C|
00841b20  e4 f1 73 e5 c1 fb 4c 0e  06 1a 3f ea 4d b8 71 66  |..s...L...?.M.qf|
00841b30  78 3c 00 00 00 00 77 25  b1 6d 00 01 89 06 c8 1a  |x<....w%.m......|
00841b40  00 00 22 84 e2 d7 3e 30  0d 8b 02 00 00 00 00 01  |.."...>0........|
00841b50  59 5a a0 fe 5f 00 00 00  00 00 86 06 60 00 00 00  |YZ.._.......`...|
00841b60  00 00 20 0e 60 00 00 00  00 00 04 80 00 00 00 00  |.. .`...........|
00841b70  6a 11 60 00 00 00 00 00  00 00 00 00 00 00 00 00  |j.`.............|
00841b80  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00843000

EDIT /lib/upgrade/fwtool.sh  -->

root@router1:~# hexdump -C /tmp/firmware.img  | tail -n 20
00841b30  78 3c 00 00 00 00 77 25  b1 6d 00 01 89 06 c8 1a  |x<....w%.m......|
00841b40  00 00 22 84 e2 d7 3e 30  0d 8b 02 00 00 00 00 01  |.."...>0........|
00841b50  59 5a a0 fe 5f 00 00 00  00 00 86 06 60 00 00 00  |YZ.._.......`...|
00841b60  00 00 20 0e 60 00 00 00  00 00 04 80 00 00 00 00  |.. .`...........|
00841b70  6a 11 60 00 00 00 00 00  00 00 00 00 00 00 00 00  |j.`.............|
00841b80  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00843000  00 00 00 00 00 00 00 00  7b 20 20 22 73 75 70 70  |........{  "supp|
00843010  6f 72 74 65 64 5f 64 65  76 69 63 65 73 22 3a 5b  |orted_devices":[|
00843020  22 6e 65 74 67 65 61 72  2c 72 37 38 30 30 22 2c  |"netgear,r7800",|
00843030  22 72 37 38 30 30 22 5d  2c 20 22 76 65 72 73 69  |"r7800"], "versi|
00843040  6f 6e 22 3a 20 7b 20 22  64 69 73 74 22 3a 20 22  |on": { "dist": "|
00843050  4f 70 65 6e 57 72 74 22  2c 20 22 76 65 72 73 69  |OpenWrt", "versi|
00843060  6f 6e 22 3a 20 22 31 38  2e 30 36 2d 53 4e 41 50  |on": "18.06-SNAP|
00843070  53 48 4f 54 22 2c 20 22  72 65 76 69 73 69 6f 6e  |SHOT", "revision|
00843080  22 3a 20 22 72 37 34 30  35 2d 30 36 61 32 30 61  |": "r7405-06a20a|
00843090  66 62 33 34 22 2c 20 22  62 6f 61 72 64 22 3a 20  |fb34", "board": |
008430a0  22 69 70 71 38 30 36 78  22 20 7d 20 7d 0a 46 57  |"ipq806x" } }.FW|
008430b0  78 30 e4 e2 56 1c 01 00  00 00 00 00 00 be        |x0..V.........|
008430be
root@router1:~#

```

jow-, 2018-12-02:
Great find, I'll drop the offending commit. On the positive side this is no random file corruption but a clean removal of the metadata trailer, so it should not result in bricks.

jow-, 2018-12-02:
I reverted the offending commit in the OpenWrt tree.

closed 2018-12-02 by jow-

hnyman, 2018-12-02:
Thanks. Good explanation from you in the fix commit.

misterti, 2019-01-28:
I am getting an error in luci for my own snapshot build, where the image is not stripped, but it doesn't seem to contain metadata at all. Is this normal for own builds?

hnyman, 2019-01-28:
The existence of metadata depends on your router's target.

Which router? Which error? 
Trying to sysupgrade from which firmware to which firmware?

misterti, 2019-01-28:
fwtool -i -q, called from fwtool.sh script, fails with "Image metadata not found" on a snapshot built image. The image itself works fine after forced flashing is selected in luci. it was just quite confusing, whether it was safe to flash it, until i found this thread. I don't know why metadata is not found in my built image, or if this is normal for snapshot builds and fwtool should consider the scenario when metadata is not there.

Asus RT-N56U
OpenWrt SNAPSHOT r9137-0465e41e05 / LuCI Master (git-19.025.67551-84ff909)