openwrt/luci #2352: Firmware upgrade by tar.gz

khareanudeep, 2018-12-05:
[image]

Hi Team,
I want to upgrade new firmware in my Board by Single Image its working file But our Requirment is Upgrade by tar.gz. Please Help me on this i also attached screenshot. please reply ASAP.
thanks.

khareanudeep, 2018-12-06:
Hi Openwrt Team ,
please help on this .

hnyman, 2018-12-06:
There is no magic in the file format being .tar.gz or not.
There is a target-specific check, if the image is in the expected format. You have not told any details about the target (ar71xx, ipq806x, ... etc.) or about the expectd format vs. your format.

You will likely need to modify the logic for the expected file format for your target in the source code.
 

But this is not a LuCI GUI issue. It simply uses the half-way target-dependent sysupgrade scripts to check the image and to flash it.

closed 2018-12-06 by hnyman

khareanudeep, 2018-12-07:
Hi hnyman thanks for reply,
yes my target is ipq806x , and i want to upgrade vai tar insidethe  tar my image is present.
suggest some solution.