openwrt/luci #2358: config ifstate in /etc/config/luci breaks LuCI

MartijnRas, 2018-12-09:
I'd not logged into the web inteface for some time, I connect over SSH once every while and install updates. So I don't know which update caused this problem, but where I expected the LuCI login page I was presented with the following error:

/usr/lib/lua/luci/dispatcher.lua:230: /etc/config/luci seems to be corrupt, unable to find section 'main'
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:230: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:127: in function </usr/lib/lua/luci/dispatcher.lua:126>

After commenting out most of the /etc/config/luci file things started sort of working again.
Ultimately everything is working as expected, however the following section is the culprit: 

config ifstate
      option interface guest-wifi        
      option ifname radio0.network2 radio1.network2        
      option bridge true

Is there a new syntax for this part, or should it be moved to another configuration file?

cshoredaniel, 2018-12-09:
What is ``config ifstate``?  I've never heard of that config section type.

jow-, 2018-12-09:
Your `option ifname` lacks quotes.

closed 2018-12-09 by jow-