openwrt/luci #2360: Migrate documentation/*.md to wiki

stokito, 2018-12-09:
In the /documentation folder are present some markdown files with HowTos. The same information is stored on luci wiki. There is some divergences like something is converted from Wiki format to Markdown and something isn't. Also in some places information seems more old that in other source.
So I tried to synchronize them and now Wiki contains latest version.
Please review those articles and remove the documentation from sources to avoid collisions.

feckert, 2019-08-11:
@jow- I think the configuration should be maintained in the source. And not in a separate system.
@stokito Thanks for working to maintain this in the Wiki
It's a matter of time until the Wiki is no longer synchronous. A Solution would be that we update the feature CI/CD to generate the doku and push this to a proper place.

stokito, 2019-08-11:
Technically speaking WIki is also can be accessed as a git repo. You can clone it like `git clone https://github.com/openwrt/luci.wiki`.
I agree that GitHub's WIki has few problems:
1. Nobody ever looking into the WIki. It should be some link in README that refer to the WIKI page i.e. `Read the Wiki https://github.com/openwrt/luci/wiki`.
2. When you cloned sources that would be nice to have docs or examples just in place. When you searching for something you may find the doc.
3. When you changing or renaming something in sources then the IDE (like CLion) can show you that you probably should rename this in documentation.

As a pros for the WIki:
1. It's easy to change it: I updated those docs without sending any PRs
2. The docs are moved out from sources and this makes cloning the repo a little bit faster especially for CI.
3. It becomes searchable by Google and other crawlers and anybody can read it directly from website.

mwoolweaver, 2019-08-11:
would it be possible to include the wiki as a submodule in a folder named docs? Or would that be to clumsy to keep updated?

stokito, 2019-08-11:
yes, that's possible and I guess that's will be a good solution

mwoolweaver, 2019-08-11:
The only con (if that's what you wanna call it) I can for see is that documentation would have to be edited via the wiki?

stokito, 2019-08-11:
I tried but I got 403 error and it looks like only collaborators can do pushes to master.
Also it looks like that you can't create PRs for Wiki. This is a big limitation.

PR #6513 (2023-08-11): docs: Synchronize with Wiki
The Wiki on GitHub has newer version of documentation. Still we may have references to the docs folder.
Update all pages but also add a notice:

> See [online wiki](https://github.com/openwrt/luci/wiki/) for latest version.

The LAR page is removed because there no any mentions of it anywhere in Luci.

This closes #2360
commit 155d2bdd (2023-08-11): docs: Synchronize with Wiki

The Wiki on GitHub has newer version of documentation.
Still we may have references to the docs folder.
Update all pages but also add a notice:

See [online wiki](https://github.com/openwrt/luci/wiki/) for latest version.

The LAR page is removed because there no any mentions of it anywhere in Luci.

This closes #2360
closed 2023-08-11 by jow-