openwrt/luci #2373: New location annotations in the po files makes the diff unreadable

feckert, 2018-12-13:
As mentioned in #2372 wouldn't it not be better to write a small tool that has the **msgid** as a passing argument? This tool then searches the sources for the corresponding line and outputs the same information.

jow-, 2018-12-13:
The change was made deliberately to give translations editors more source code context to work with.

closed 2018-12-13 by jow-

feckert, 2018-12-13:
@jow If the translation editors need more source code context during translation time and using the tool to get the translation is not a option then I have another proposal.

During translation work we execute the tool to add more context and if the work is done (before commit) we could the remove the context.

The advantage is that the **diffs** remain more transparent and also the data volume for the **diffs** decreases.

jow-, 2018-12-13:
This is not the typical workflow used by a translation editor, people use programs like `poedit` to modify the files. If you want diffs to remain readable, simply split the `*.po` / `*.pot` updates and the translation updates into separate commits.

feckert, 2018-12-13:
Convinced thanks