openwrt/luci #2379: luci-app-lxc missing vital dependcies for basic operations

n8v8R, 2018-12-18:
@dibdot @ryu0 

such as `kmod-veth` `lxc-start` `lxc-stop` `lxc-attach`

also the `PKG_MAINTAINER:=Petar Koretic <petar.koretic@sartura.hr>` does not appear to active on GitHub or the repo?

ryu0, 2018-12-18:
It depends on lxc-attach already. lxc-start/lxc-stop appears to be irrelevant, as the LUCI code makes calls to ubus when it is asked to start/stop a container. Why would it need them?

n8v8R, 2018-12-18:
> lxc-start/lxc-stop appears to be irrelevant, as the LUCI code makes calls to ubus when it is asked to start/stop a container.

Wasn't aware of it.

---
> It depends on lxc-attach already

That I apparently missed.

---
leaves `kmod-veth`. Without it the bridge with the host cannot get connected.

dibdot, 2018-12-20:
'kmod-veth' is optional, lxc containers & the luci app does not need this dependency. Anyway, the luci app informs the user about this requirement for network connectivity.

closed 2018-12-20 by dibdot