openwrt/luci #2388: Wireless status page format - cosmetic, but easy to fix ?

wapebira, 2018-12-21:
The Wireless Status - Wireless Overview page creates a table for each interface, which will cause the text and the action buttons not being aligned. It should use only a single "cbi-section-node" with a single table for all the interfaces.
[https://imgur.com/kfi4w2K](https://imgur.com/kfi4w2K)
`<div class="cbi-map" id="cbi-wireless">
  <h2 name="content">Wireless Overview</h2>
  
    <div class="cbi-section-node">
      <div class="table">
          <div class="tr cbi-rowstyle-2">
          ...
          <div class="tr cbi-rowstyle-1">
          ...
      </div>
    </div>

    <div class="cbi-section-node">
      <div class="table">
          <div class="tr cbi-rowstyle-2">
          ...
          <div class="tr cbi-rowstyle-1">
          ...
      </div>
    </div>

    <div class="cbi-section-node">
      <div class="table">
          <div class="tr cbi-rowstyle-2">
          ...
          <div class="tr cbi-rowstyle-1">
          ...
      </div>
    </div>
  </div>
</div>`

should look more like this [https://imgur.com/qPD655w](https://imgur.com/qPD655w) 
`<div class="cbi-map" id="cbi-wireless">
  <h2 name="content">Wireless Overview</h2>
  
    <div class="cbi-section-node">
      <div class="table">
          <div class="tr cbi-rowstyle-2">
          ...
          <div class="tr cbi-rowstyle-1">
          ...
          <div class="tr cbi-rowstyle-2">
          ...
          <div class="tr cbi-rowstyle-1">
          ...
           <div class="tr cbi-rowstyle-2">
          ...
          <div class="tr cbi-rowstyle-1">
          ...
  </div>
</div>`

jow-, 2018-12-29:
The markup was intentionally be written this way to allow the layout to flex when needed. I do not consider this important enough to fix it.

closed 2018-12-29 by jow-

wapebira, 2018-12-30:
I did not say is important either... and flex it is not a problem in this case. The problem is the template.
Here is the fix if you want to use it

Step 1: In  wifi_overview.lua

add before this loop: `for _, dev in ipairs(ntm:get_wifidevs())`

`s = m:section(NamedSection, "__wifioverview__") function s.render(self,sid) tpl.render_string([[
		<div class="cbi-section-node"><div class="table">	]]) end`

Step 2: before `s = m:section(NamedSection, "__assoclist__")` add

`function s.render(self, sid) tpl.render_string([[</div></div>	]]) end`

Step 3: in wifi_overview.htm remove these <div> sections
`<div class="cbi-section-node">
	<div class="table">`