openwrt/luci #2398: cbi.js: dynlist and password input initialized more than once

yousong, 2018-12-26:
The issue was observed on luci-app-shadowsocks-libev, but it's likely a regression during code refactor and can also affect other components.

The 1st issue being that every dynlist was initialised 3 times on "Redir Rules" tab.  This was caused by the fact that there are there `cbi_init()` call in the generated html.  The other two are triggered by filebrowser.

[image]

The 2nd issue being that it seems for every "password = true" field, there are two dom input object present.  One generated directly in the html response.  The other initialised by js code.  The effect can be observed on the "Remote Servers" tab.

[image]

yousong, 2018-12-29:
I understand.  Previously browser autocomplete can indeed cause trouble.  But it seems when embedded in a table, the added input element will be visible, despite the css code `left: -1000px`

[image]

Another minor undesirable change is that previously the `Reveal/hide password` button stick to the input element on the same row, not wrapping to the next line.

[image]

jow-, 2018-12-29:
Will look into this asap.