openwrt/luci #2405: dhcp redirecting client to tftpd running on other machine in lan

mistepien, 2018-12-28:
> config  dhcp 'lan'
	option interface 'lan'
	option start '100'
	option limit '150'
	option leasetime '12h'
	list dhcp_option 'option:mtu, 9000'
	list dhcp-option '66,"192.168.1.8'
	list dhcp_option 'option:bootfile-name, `pxelinux.0'

if I put it into /etc/config/dhcp then luci related to dhcp goes south.
It would be great to have ability to configure it via luci.

cshoredaniel, 2018-12-28:
@mistepien You have an extraneous quote (") in ``list dhcp-option '66,"192.168.1.8'``

jow-, 2018-12-29:
Not LuCI related.

closed 2018-12-29 by jow-

mistepien, 2019-01-05:
``config dhcp 'lan'
	option interface 'lan'
	option start '100'
	option limit '150'
	option leasetime '12h'
	list dhcp_option 'option:mtu, 9000'
	list dhcp-option '66,192.168.1.11'
	list dhcp_option 'option:bootfile-name, pxelinux.0'

mistepien, 2019-01-05:
ok, you were right.
this is a cli solution:
`uci add_list dhcp.@dnsmasq[0].dhcp_boot='pxelinux.0,<name of external tftp pxe server>,<IP of external tftp pxe server'`