openwrt/luci #2411: Wifi country value per interface always returns global

wapebira, 2018-12-31:
For example:

uci show wireless
```
wireless.radio0.country='US'
...
wireless.radio1.country='US'
...
wireless.radio2.country='US'`
```

iw reg get
```
global
country 98: DFS-FCC
...
phy#2
country US: DFS-FCC
...
phy#1
country US: DFS-FCC
...
phy#0
country US: DFS-FCC`
```

LuCi always showing "Country: 98"
[image]

jow-, 2018-12-31:
This is intentional. While the `option country` setting in OpenWrt is defined per-radio, there is only one global regulatory parameter in mac80211. When we'd change iwinfo to display the per-phy country code, it would never reflect what has been configured but always show the hard-wired eeprom locality of the radio phy.

wapebira, 2019-01-02:
After digging more into this, the issue is with mwlwifi driver.
The Marvell 88W8964 has the country and power tables burned into it. 88W8964 uses mwlwifi driver.
WRT32X has another chip 88W8887 which uses another driver and will cause the mwlwifi to set the wrong country code to 98.
mwlwifi it is a sup par implementation. I cannot even get it work on 160mhz... which works on stock
After removing the phy2, luci shows get the country correctly.

Sorry for trouble jow

closed 2019-01-04 by jow-

serf342, 2019-03-03:
@wapebira i am having the exact same issue with the WRT1900ACS router.
When i flashed the factory image 18.06.2 over the linksys, i go to wireless in LUCi and change country to Taiwan, then try change the transmit power to 30dbm and it still only goes to 20dbm.
When i checked in terminal CLI, i saw these results:

```
root@OpenWrt:~# iw reg get
global
country 98: DFS-FCC
...
phy#1
country CN: DFS-FCC
...
phy#0
country CN: DFS-FCC
...
```

The country never changes, even when doing `iw reg set TW`

How exactly did you solve this? How did you remove phy2 ? In my case what phy should i remove?

on WRT1200AC routers this was never a problem, why is it a problem on WRT1900ACS ?
Please help

wapebira, 2019-03-03:
@serf342 
Not really sure about WRT1900ACS. I don't have one and don't have time to look into what wifi hardware it has. But on a WRT32, I removed the phy2 like this

`opkg remove kmod-mwifiex-sdio`
`opkg remove mwifiex-sdio-firmware`
`reboot`

You could maybe find more info here
[https://github.com/kaloz/mwlwifi/blob/c2f82f1ac18ed7fb48be1eb8d678a3cf32192069/README.md](url)