openwrt/luci #2414: dropbear authentication succeeded but terminates immeadiately

amworsley, 2019-01-02:
I have added an ssh key and enabled ssh but it always terminates with end of input:

Version is: OpenWrt 18.06.1 r7258-5eb055306f / LuCI openwrt-18.06 branch (git-18.228.31946-f64b152) 

...
debug1: Server accepts key: /home/amw/.ssh/id_rsa RSA SHA256:PT0K.... agent
debug1: Authentication succeeded (publickey).
Authenticated to 10.1.0.8 ([10.1.0.8]:22).
debug1: channel 0: new [client-session]
debug2: channel 0: send open
debug1: Entering interactive session.
debug1: pledge: network
debug2: channel_input_open_confirmation: channel 0: callback start
debug2: fd 3 setting TCP_NODELAY
debug2: client_session2_setup: id 0
debug2: channel 0: request pty-req confirm 1
debug1: Sending environment.
debug1: Sending env LANG = en_AU
debug2: channel 0: request env confirm 0
debug2: channel 0: request shell confirm 1
debug2: channel_input_open_confirmation: channel 0: callback done
debug2: channel 0: open confirm rwindow 24576 rmax 32759
debug2: channel_input_status_confirm: type 99 id 0
debug2: PTY allocation request accepted on channel 0
debug2: channel_input_status_confirm: type 99 id 0
debug2: shell request accepted on channel 0
debug2: channel 0: rcvd eof
...

Similarly trying to run a command: ssh -vv root@10.1.0.8 /bin/ls /
....
debug1: Authentication succeeded (publickey).
Authenticated to 10.1.0.8 ([10.1.0.8]:22).
debug1: channel 0: new [client-session]
debug2: channel 0: send open
debug1: Entering interactive session.
debug1: pledge: network
debug2: channel_input_open_confirmation: channel 0: callback start
debug2: fd 3 setting TCP_NODELAY
debug2: client_session2_setup: id 0
debug1: Sending environment.
debug1: Sending env LANG = en_AU
debug2: channel 0: request env confirm 0
debug1: Sending command: /bin/ls /
debug2: channel 0: request exec confirm 1
debug2: channel_input_open_confirmation: channel 0: callback done
debug2: channel 0: open confirm rwindow 24576 rmax 32759
debug2: channel_input_status_confirm: type 99 id 0
debug2: exec request accepted on channel 0
debug2: channel 0: rcvd eof
debug2: channel 0: output open -> drain
debug2: channel 0: obuf empty
debug2: channel 0: chan_shutdown_write (i0 o1 sock -1 wfd 6 efd 7 [write])
debug2: channel 0: output drain -> closed
debug1: client_input_channel_req: channel 0 rtype exit-status reply 0
debug2: channel 0: rcvd close
....
I do see these two entries in the logs:
...
Wed Jan  2 13:07:21 2019 authpriv.notice dropbear[16416]: Pubkey auth succeeded for 'root' with key sha1!! 7f:61:18:7b:36:fd:59:d8:bd:77:41:d3:33:78:29:7c:12:65:42:fd from 10.1.0.122:42742
Wed Jan  2 13:07:21 2019 authpriv.info dropbear[16416]: Exit (root): Disconnect received

cshoredaniel, 2019-01-02:
Have you tried running debug on dropbear?  Usually this indicates something is not right on the server side (or the server doesn't like something from the client).

jow-, 2019-01-02:
This is not a LuCI issue.

closed 2019-01-02 by jow-

amworsley, 2019-01-03:
On Wed, 2 Jan 2019 at 19:55, Daniel Dickinson <notifications@github.com>
wrote:

> Have you tried running debug on dropbear? Usually this indicates something
> is not right on the server side (or the server doesn't like something from
> the client).
>
How do I enabled debug on dropbear *without* access to an ssh server ?

If this is not a LuCI issue - where does the issue belong?

amworsley, 2019-01-03:
On Thu, 3 Jan 2019 at 14:26, Daniel Dickinson <notifications@github.com>
wrote:

> If you had a) serial b) enable password authentication through LuCI
> (assuming you have it installed), and start dropbear with debug on another
> port and connect to that port from your client.
>
Unfortunately it is in use and has the "lid on"
I was wondering if there was an established way to open up a raw port.
e.g. netcat via the page which lists scripts to run on start up.

Or a simple debug flag setting.
Not having a very good feel for the system it would be very error prone to
guess and try various things.

Thanks

Andrew

cshoredaniel, 2019-01-03:
Since you have LuCI you could try the second suggestion...use a password based login via SSH (assuming the problem is keys) and run ``dropbear -F -E -p 2222``

I'm wondering if you're actually having a out of flash issue (can you successfully change and apply settings that survive reboot)?

amworsley, 2019-01-04:
Ok I'm not sure if this is what you intended but I added:
`dropbear -F -E -p 2222 &`

to the System/Startup - Local Startup section and it appears to run dropbear on that port but it fails exactly as before - the public/private keys seem find - the shell that is started just immediately terminates.
`ssh -vv -p2222 root@10.8.1.1`
yields:
```....
debug1: Authentication succeeded (publickey).
Authenticated to 10.8.1.1 ([10.8.1.1]:2222).
debug1: channel 0: new [client-session]
debug2: channel 0: send open
debug1: Entering interactive session.
debug1: pledge: network
debug2: channel_input_open_confirmation: channel 0: callback start
debug2: fd 3 setting TCP_NODELAY
debug2: client_session2_setup: id 0
debug2: channel 0: request pty-req confirm 1
debug1: Sending environment.
debug1: Sending env LANG = en_AU
debug2: channel 0: request env confirm 0
debug2: channel 0: request shell confirm 1
debug2: channel_input_open_confirmation: channel 0: callback done
debug2: channel 0: open confirm rwindow 24576 rmax 32759
debug2: channel_input_status_confirm: type 99 id 0
debug2: PTY allocation request accepted on channel 0
debug2: channel_input_status_confirm: type 99 id 0
debug2: shell request accepted on channel 0
[2065] Jan 04 12:03:26 Exit (root): Child failed
debug2: channel 0: rcvd eof
debug2: channel 0: output open -> drain
debug2: channel 0: obuf empty
debug2: channel 0: chan_shutdown_write (i0 o1 sock -1 wfd 6 efd 7 [write])
debug2: channel 0: output drain -> closed
debug1: client_input_channel_req: channel 0 rtype exit-status reply 0
debug2: channel 0: rcvd close
debug2: channel 0: chan_shutdown_read (i0 o3 sock -1 wfd 5 efd 7 [write])
debug2: channel 0: input open -> closed
debug2: channel 0: almost dead
debug2: channel 0: gc: notify user
debug2: channel 0: gc: user detached
debug2: channel 0: send close
debug2: channel 0: is dead
debug2: channel 0: garbage collecting
debug1: channel 0: free: client-session, nchannels 1
Connection to 10.8.1.1 closed.
Transferred: sent 3024, received 1536 bytes, in 0.0 seconds
Bytes per second: sent 272996.0, received 138664.6
debug1: Exit status 1
```
In the logs:
```
Fri Jan  4 12:11:25 2019 authpriv.info dropbear[2286]: Child connection from 10.8.1.81:56826
Fri Jan  4 12:11:25 2019 daemon.notice hostapd: Station e0:ca:94:e7:ae:60 not allowed to authenticate
Fri Jan  4 12:11:25 2019 daemon.notice hostapd: Station e0:ca:94:e7:ae:60 not allowed to authenticate
Fri Jan  4 12:11:26 2019 authpriv.notice dropbear[2286]: Pubkey auth succeeded for 'root' with key sha1!! 7f:61:18:7b:36:fd:59:d8:bd:77:41:d3:33:78:29:7c:12:65:42:fd from 10.8.1.81:56826
Fri Jan  4 12:11:26 2019 authpriv.info dropbear[2286]: Exit (root): Disconnect received
```

amworsley, 2019-01-04:
Ok - installed **ncat** package and added this entry to startup
'ncat --exec "/bin/sh" -l 8081 --keep-open &'

Then can log into a shell (no prompt but runs commands) with
`ncat 10.8.1.1 8081`

  * This shows shell is there and daemon is running - but no clues to reason for failure
  * Installed **strace** package and run :
`strace -f -s128 -o /tmp/trace.txt /usr/sbin/dropbear -F -p 5555 &`

  * We get the following:
````
2277  execve("/bin/zsh", ["-zsh"], 0x43ae90 /* 9 vars */) = -1 ENOENT (No such file or directory)
2277  clock_gettime(CLOCK_REALTIME, {tv_sec=1546567263, tv_nsec=631379551}) = 0
2277  getpid()                          = 2277
2277  sendto(10, "<86>Jan  4 02:01:03 dropbear[2277]: Exit (root): Child failed\n", 62, 0, NULL, 0) = -1 EBADF (Bad file descriptor)
2277  close(-2)                         = -1 EBADF (Bad file descriptor)
2277  close(-2)                         = -1 EBADF (Bad file descriptor)
2277  open("/etc/passwd", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
2277  fcntl64(3, F_SETFD, FD_CLOEXEC)   = 0
2277  fcntl64(3, F_SETFD, FD_CLOEXEC)   = 0
2277  readv(3, [{iov_base="", iov_len=0}, {iov_base="root:x:0:0:root:/root:/bin/zsh\ndaemon:*:1:1:daemon:/var:/bin/false\nftp:*:55:55:ftp:/home/ftp:/bin/false\nnetwork:*:101:101:networ"..., iov_len=1024}], 2) = 244
2277  _llseek(3, -213, [31], SEEK_CUR)  = 0
2277  close(3)                          = 0
2277  geteuid()                         = 0
2277  clock_gettime(CLOCK_REALTIME, {tv_sec=1546567263, tv_nsec=633670930}) = 0
2277  chown("/dev/pts/0", 0, 0)         = 0
2277  chmod("/dev/pts/0", 0666)         = 0
2277  exit_group(1)                     = ?
2277  +++ exited with 1 +++
````
  * Yep - root is running zsh !
`cat /etc/passwd
root:x:0:0:root:/root:/bin/zsh
daemon:*:1:1:daemon:/var:/bin/false
ftp:*:55:55:ftp:/home/ftp:/bin/false
network:*:101:101:network:/var:/bin/false
nobody:*:65534:65534:nobody:/var:/bin/false
dnsmasq:x:453:453:dnsmasq:/var/run/dnsmasq:/bin/false
`
  * There is no zsh installed - so failure....
  * There appears to be enough space for zsh (1.5Mb)
'''
df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 2.5M      2.5M         0 100% /rom
tmpfs                    29.5M      1.2M     28.3M   4% /tmp
/dev/mtdblock5           11.9M    852.0K     11.0M   7% /overlay
overlayfs:/overlay       11.9M    852.0K     11.0M   7% /
tmpfs                   512.0K         0    512.0K   0% /dev

''''
  * Hmms seems to take take requests to successfully install the package...
  * But it installs it as **/usr/bin/zsh** and so it still **fails!**
  * Ok looks like /bin is modifiable so add symbolic link `ln -s /usr/bin/zsh /bin/zsh`
 *AND* Yes we can no login:
```
% ssh root@10.8.1.1
wndr3700# 
```

cshoredaniel, 2019-01-04:
Excellent -- why zsh out of curiousity?

Anyway, excellent!

cshoredaniel, 2019-01-04:
@amworsley are sure it's unmodified stock openwrt?  That's not something that's supposed to be the case with stock openwrt.  As for the /usr/bin/zsh could you file an issue on zsh?

amworsley, 2019-01-04:
I did upgrade it from an old bufferbloat version of the system but I had not modified the system - hence actually having to use ncat to access the machine.

Does an upgrade actually re-create these files?
It looks like this file dates from the Oct 2017 before the date of the OS - Aug 2018 ?

```
wndr3700# uname -a
Linux wndr3700 4.9.120 #0 Thu Aug 16 07:51:15 2018 mips GNU/Linux
wndr3700# ls -ltrh /etc/      
drwxr-xr-x    1 root     root           0 Jan  1  1970 ppp
drwxr-xr-x    1 root     root           0 Jan  1  1970 opkg
drwxr-xr-x    1 root     root           0 Jan  1  1970 luci-uploads
drwxr-xr-x    1 root     root           0 Jan  1  1970 iproute2
drwxr-xr-x    1 root     root           0 Jan  1  1970 dropbear
drwxr-xr-x    1 root     root           0 Jan  1  1970 sysctl.d
-rw-r--r--    1 root     root         913 Sep 21  2014 sysctl.conf
-rw-r--r--    1 root     root         101 Sep 21  2014 inittab
-rw-r--r--    1 root     root          20 Sep 21  2014 hosts
-rw-r--r--    1 root     root          31 Oct  1  2014 fw_env.config
-rw-r--r--    1 root     root          18 Apr 16  2016 shells
-rw-r--r--    1 root     root         487 Apr 16  2016 profile
-rw-r--r--    1 root     root         128 Oct 18  2017 sysupgrade.conf
-rw-r--r--    1 root     root        2.9K Oct 18  2017 services
-rw-r--r--    1 root     root        2.5K Oct 18  2017 protocols
-rw-r--r--    1 root     root         110 Oct 18  2017 opkg.conf
-rw-r--r--    1 root     root         352 Oct 18  2017 firewall.user
-rw-r--r--    1 root     root        1.3K Oct 18  2017 dnsmasq.conf
-rw-------    1 root     root         178 Oct 18  2017 shadow
-rw-r--r--    1 root     root         244 Oct 18  2017 passwd
-rw-r--r--    1 root     root         145 Oct 18  2017 group
lrwxrwxrwx    1 root     root          16 Aug 16 17:51 resolv.conf -> /tmp/resolv.conf
drwxr-xr-x    2 root     root         491 Aug 16 17:51 rc.d
-rwxr-xr-x    1 root     root        2.3K Aug 16 17:51 rc.common
....
-rw-r--r--    1 root     root         396 Aug 16 17:51 banner
lrwxrwxrwx    1 root     root           7 Aug 16 17:51 TZ -> /tmp/TZ
drwxr-xr-x    1 root     root           0 Aug 16 17:51 uci-defaults
-rw-r--r--    1 root     root        1.1K Aug 16 17:51 board.json
-rw-r--r--    1 root     root           0 Aug 16 17:51 ethers
-rw-------    1 root     root         512 Dec 28 15:26 urandom.seed
drwxr-xr-x    1 root     root           0 Jan  4 11:58 config
-rw-r--r--    1 root     root         202 Jan  4 13:32 rc.local
```

cshoredaniel, 2019-01-04:
>
> I did upgrade it from an old bufferbloat version of the system but I
> had not modified the system - hence actually having to use ncat to
> access the machine.
>
> Does an upgrade actually re-create these files?
Upgrade preserves config including passwd, shadow and group.  So that
explains it.

amworsley, 2019-01-04:
Ok - I'm still having a little problem.

I just changed /etc/passwd to use /bin/sh and it refuses to accept my public key?
```sed -i '/^root:/s/zsh/sh/'  /etc/passwd```
I then changed /etc/passwd back to /bin/zsh and it is ok!
```sed -i '/^root:/s/sh/zsh/'  /etc/passwd```

 * If I use ash 
```sed -i '/^root:/s/zsh/ash/'  /etc/passwd```
 * I get:
```
BusyBox v1.28.3 () built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 18.06.1, r7258-5eb055306f
 -----------------------------------------------------
%m %n %c %@(%?)%# 
```
  * It seems my **/etc/profile** has that PS1 setting.

amworsley, 2019-01-04:
* I installed **vim** and just commented out the PS1 definition - now **/bin/ash** gives me a normal shell.
  * Is there a way to reset things back to default? I have a feeling that I am going to be bitten by endless minor quirks/problems....

amworsley, 2019-01-04:
I just removed the large **zsh** package and now can login normally....

cshoredaniel, 2019-01-04:
At this point I would suggest https://forum.openwrt.org ...