openwrt/luci #2431: Cannot change anything through luci-app-rp-pppoe-server

heychirag, 2019-01-05:
No options are visible on PPPoE Server config page in luci.

[image]

dibdot, 2019-01-05:
This was already answered in the forum.

heychirag, 2019-01-05:
drop the link, please. Even Google is unable to find it.

dibdot, 2019-01-05:
It's probably your own crosspost in the forum ... https://forum.openwrt.org/t/rp-pppoe-server-not-found-even-after-a-package-list-update/28311/6

cshoredaniel, 2019-01-05:
Ah, without SSH you can't fix it -- I think that's a valid point -- I'll do an update so that the necessary sections can be created from within LuCI.  @dibdot could you reopen and assign to me so I can track it?

dibdot, 2019-01-06:
@cshoredaniel re-opened as requested.

closed 2019-01-06 by hnyman

cshoredaniel, 2019-01-06:
@heychirag should work in master now.

PR #2434 (2019-01-06): luci-app-rp-pppoe-server: Add ability to create uci sections
Currently if no non-commented uci sections exist in the config then
it's impossible to create a config from LuCI. Fix that, which also
allows what the initscript does, which is multiple instances. While
we're at it create the translation template.

Closes: #2431

Run tested on Raspberry Pi B+
commit 96cf73a5 (2019-01-06): luci-app-rp-pppoe-server: Add ability to create uci sections

Currently if no non-commented uci sections exist in the config then
it's impossible to create a config from LuCI. Fix that, which also
allows what the initscript does, which is multiple instances. While
we're at it create the translation template.

Closes: #2431