openwrt/luci #2433: Share WiFi access credential with QRcode

pgaufillet, 2019-01-06:
Hi LuCI team!

It would be great if the Wifi module of LuCI could display a QRcode for sharing Wifi credentials.
See [Qifi](https://github.com/evgeni/qifi) for example.

Thanks for your great job!

leonghui, 2019-01-15:
Interesting suggestion. Instead of Javascript, [qrencode ](https://github.com/openwrt/packages/tree/master/libs/qrencode) is currently used by 3 LuCI packages to generate QR codes:
- luci-app-wireguard
- luci-app-fwknopd
- luci-app-travelmate

greyltc, 2019-03-17:
This would be great. Android Q and above has joining wifi via qr code baked into the os in a prominent way.

feckert, 2019-04-23:
What format/config file should the QR-Code have?

**WIFI:S:\<ssid\>;T:\<encryption\>;P:\<password\>;H:\<hide\>**

plm, 2019-12-31:
Travelmate has an example of [the `qrencode` arguments](https://github.com/openwrt/luci/blob/29e4b397da2126ef7a433ffb8dde47d879a47ad6/applications/luci-app-travelmate/luasrc/view/travelmate/runtime.htm#L200) required to generate a QR code for the Wifi credentials.

hmh, 2021-03-02:
> This would be great. Android Q and above has joining wifi via qr code baked into the os in a prominent way.

Also iOS 11+, in the built-in camera application.

OW87, 2022-02-13:
I would also like to see this this feature for the Wifi networks.

If we could add a QR Code on the line for every wifi like this :
[image]

@feckert : Are you working on this feature ? 
Thank you for your reply.

jonathan-annett, 2022-03-19:
If it helps the luci developers, here is a simple page i wrote for creating QR codes (it uses a few libs - it's pure javascript, happens client side in browser )

**you might need to host these on the router so i'm not sure if size constraints prevent using this approach.** 

I don't suggest you use it like this for security reasons but a working sample QR can be generated by appending a string to the url ```https://qr.1mb.site/?code=``` :  [sample link](https://qr.1mb.site/?code=WIFI:T:WPA;S:myssid;P:mysecret;;)
 
To implement it in the luci ui, have a look at this source https://qr.1mb.site/index.js (it parses the passed in url as a string and displays it verbatim. wifi QR codes have a specific syntax, which you can see after "code=" in the url - which for simplity in this demo isn't urlencoded. as you'll be generating it locally in the browser url encoding is irrelevant, i mention it here as obviously complex passwords can't be demoed via this generic site without url ecoding)

the specific syntax for the qr to pass into the library function is documented [here](https://pocketables.com/2022/01/how-to-format-that-wifi-qr-code-in-plain-text.html) however for a wpa2 protected ssid it is basically ```WIFI:S:ssid;T:WPA;P:password;;```

 
in a nutshell, i'm pulling in a few files via a cdn:  

```https://cdnjs.cloudflare.com/ajax/libs/qrcodejs/1.0.0/qrcode.min.js```  - [link](https://cdnjs.cloudflare.com/ajax/libs/qrcodejs/1.0.0/qrcode.min.js) - [info](https://cdnjs.com/libraries/qrcodejs) - [github](https://github.com/davidshimjs/qrcodejs?utm_source=cdnjs&utm_medium=cdnjs_link&utm_campaign=cdnjs_library)

```https://cdnjs.cloudflare.com/ajax/libs/jquery/1.8.3/jquery.min.js``` - [link](https://cdnjs.cloudflare.com/ajax/libs/jquery/1.8.3/jquery.min.js) - library used by qrcode - [info](https://cdnjs.com/libraries/jquery/1.8.3)

and calling them from index.js

hmh, 2022-03-21:
fwknock support already generates QRcodes from within Luci, whatever is done should be shared between the two packages for space savings.

systemcrash, 2025-11-26:
Closed by 805b2db67012ee76ea8baf57cf83c4dc31b72832

closed 2025-11-26 by systemcrash