openwrt/luci #2453: Update Lists - Failed to establish connection

davidc502, 2019-01-13:
This has been going on for quite some time, and apologize if already reported --

When updating lists in LuCi (Doesn't matter the theme), the following error occurs. 

[code]
Executing package manager
Downloading https://davidc502sis.dynamic-dns.net/snapshots/r9028/targets/mvebu/cortexa9/packages/Packages.gz
*** Failed to download the package list from https://davidc502sis.dynamic-dns.net/snapshots/r9028/targets/mvebu/cortexa9/packages/Packages.gz

Downloading https://davidc502sis.dynamic-dns.net/snapshots/r9028/packages/arm_cortex-a9_vfpv3/base/Packages.gz
*** Failed to download the package list from https://davidc502sis.dynamic-dns.net/snapshots/r9028/packages/arm_cortex-a9_vfpv3/base/Packages.gz

Downloading https://davidc502sis.dynamic-dns.net/snapshots/r9028/packages/arm_cortex-a9_vfpv3/darkmatter/Packages.gz
*** Failed to download the package list from https://davidc502sis.dynamic-dns.net/snapshots/r9028/packages/arm_cortex-a9_vfpv3/darkmatter/Packages.gz

Downloading https://davidc502sis.dynamic-dns.net/snapshots/r9028/packages/arm_cortex-a9_vfpv3/luci/Packages.gz
*** Failed to download the package list from https://davidc502sis.dynamic-dns.net/snapshots/r9028/packages/arm_cortex-a9_vfpv3/luci/Packages.gz

Downloading https://davidc502sis.dynamic-dns.net/snapshots/r9028/packages/arm_cortex-a9_vfpv3/packages/Packages.gz
*** Failed to download the package list from https://davidc502sis.dynamic-dns.net/snapshots/r9028/packages/arm_cortex-a9_vfpv3/packages/Packages.gz

Downloading https://davidc502sis.dynamic-dns.net/snapshots/r9028/packages/arm_cortex-a9_vfpv3/routing/Packages.gz
*** Failed to download the package list from https://davidc502sis.dynamic-dns.net/snapshots/r9028/packages/arm_cortex-a9_vfpv3/routing/Packages.gz

Downloading https://davidc502sis.dynamic-dns.net/snapshots/r9028/packages/arm_cortex-a9_vfpv3/telephony/Packages.gz
*** Failed to download the package list from https://davidc502sis.dynamic-dns.net/snapshots/r9028/packages/arm_cortex-a9_vfpv3/telephony/Packages.gz

Errors
Failed to establish connection
Failed to establish connection
Failed to establish connection
Failed to establish connection
Failed to establish connection
Failed to establish connection
Failed to establish connection
Collected errors:
 * opkg_download: Failed to download https://davidc502sis.dynamic-dns.net/snapshots/r9028/targets/mvebu/cortexa9/packages/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download https://davidc502sis.dynamic-dns.net/snapshots/r9028/packages/arm_cortex-a9_vfpv3/base/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download https://davidc502sis.dynamic-dns.net/snapshots/r9028/packages/arm_cortex-a9_vfpv3/darkmatter/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download https://davidc502sis.dynamic-dns.net/snapshots/r9028/packages/arm_cortex-a9_vfpv3/luci/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download https://davidc502sis.dynamic-dns.net/snapshots/r9028/packages/arm_cortex-a9_vfpv3/packages/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download https://davidc502sis.dynamic-dns.net/snapshots/r9028/packages/arm_cortex-a9_vfpv3/routing/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download https://davidc502sis.dynamic-dns.net/snapshots/r9028/packages/arm_cortex-a9_vfpv3/telephony/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

The opkg update command failed with code 7.
``
HOWEVER -- running opkg update from command line works as expected.
[code]
root@lede:~# opkg update
Downloading https://davidc502sis.dynamic-dns.net/snapshots/r9028/targets/mvebu/cortexa9/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/lede_core
Downloading https://davidc502sis.dynamic-dns.net/snapshots/r9028/targets/mvebu/cortexa9/packages/Packages.sig
Signature check passed.
Downloading https://davidc502sis.dynamic-dns.net/snapshots/r9028/packages/arm_cortex-a9_vfpv3/base/Packages.gz
Updated list of available packages in /var/opkg-lists/lede_base
Downloading https://davidc502sis.dynamic-dns.net/snapshots/r9028/packages/arm_cortex-a9_vfpv3/base/Packages.sig
Signature check passed.
Downloading https://davidc502sis.dynamic-dns.net/snapshots/r9028/packages/arm_cortex-a9_vfpv3/darkmatter/Packages.gz
Updated list of available packages in /var/opkg-lists/lede_darkmatter
Downloading https://davidc502sis.dynamic-dns.net/snapshots/r9028/packages/arm_cortex-a9_vfpv3/darkmatter/Packages.sig
Signature check passed.
Downloading https://davidc502sis.dynamic-dns.net/snapshots/r9028/packages/arm_cortex-a9_vfpv3/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/lede_luci
Downloading https://davidc502sis.dynamic-dns.net/snapshots/r9028/packages/arm_cortex-a9_vfpv3/luci/Packages.sig
Signature check passed.
Downloading https://davidc502sis.dynamic-dns.net/snapshots/r9028/packages/arm_cortex-a9_vfpv3/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/lede_packages
Downloading https://davidc502sis.dynamic-dns.net/snapshots/r9028/packages/arm_cortex-a9_vfpv3/packages/Packages.sig
Signature check passed.
Downloading https://davidc502sis.dynamic-dns.net/snapshots/r9028/packages/arm_cortex-a9_vfpv3/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/lede_routing
Downloading https://davidc502sis.dynamic-dns.net/snapshots/r9028/packages/arm_cortex-a9_vfpv3/routing/Packages.sig
Signature check passed.
Downloading https://davidc502sis.dynamic-dns.net/snapshots/r9028/packages/arm_cortex-a9_vfpv3/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/lede_telephony
Downloading https://davidc502sis.dynamic-dns.net/snapshots/r9028/packages/arm_cortex-a9_vfpv3/telephony/Packages.sig
Signature check passed.
[/code]

jow-, 2019-01-15:
Maybe a different wget without SSL support is called when invoked through uhttpd, in any case this is not a LuCI issue.

closed 2019-01-15 by jow-

davidc502, 2019-01-16:
Interesting.

miguel-orange, 2019-02-26:
@davidc502 
I believe it's a cert mismatch.  `davidc502sis.dynamic-dns.net` attempts to use a cert for `dc502wrt.org`. Changing `opkg/distfeeds.conf` to use `dc502wrt.org` worked for me.
```
src/gz lede_core https://dc502wrt.org/snapshots/r9287/targets/mvebu/cortexa9/packages
src/gz lede_base https://dc502wrt.org/snapshots/r9287/packages/arm_cortex-a9_vfpv3/base
src/gz lede_darkmatter https://dc502wrt.org/snapshots/r9287/packages/arm_cortex-a9_vfpv3/darkmatter
src/gz lede_luci https://dc502wrt.org/snapshots/r9287/packages/arm_cortex-a9_vfpv3/luci
src/gz lede_packages https://dc502wrt.org/snapshots/r9287/packages/arm_cortex-a9_vfpv3/packages
src/gz lede_routing https://dc502wrt.org/snapshots/r9287/packages/arm_cortex-a9_vfpv3/routing
src/gz lede_telephony https://dc502wrt.org/snapshots/r9287/packages/arm_cortex-a9_vfpv3/telephony

```

ferboiar, 2021-05-23:
```
root@OpenWrt:~#  * opkg_download: Failed to download https://dc502wrt.org/snapshots/r11829/packages/
arm_cortex-a9_vfpv3/telephony/Packages.gz, wget returned 4.

root@OpenWrt:~#  *
```
is the repo up?