openwrt/luci #2472: Can you support linksys EA6350v3 ？

cheneylaw, 2019-01-21:
luci-app-advanced-reboot can't recognition  linksys EA6350v3 dual boot，Can  support it？THX.

hnyman, 2019-01-21:
> luci-app-advanced-reboot can't recognition linksys EA6350v3 dual boot，Can support it？THX.

@stangri

stangri, 2019-01-22:
@cheneylaw -- can you please supply any relevant information? Like a link to the ToH page, links to relevant commits, etc? Do you have this device? Will you be able to test the package after the support is added? Do you know how to switch between partitions from CLI?

hnyman, 2019-01-27:
The support for EA6350v3 has been merged to the main openwrt repo.
See the most recent commits before this:
https://github.com/openwrt/openwrt/commits/a873b292840848b67addb5be0c762d292bde33a1

The factory image seems to need more complex checksum logic for the image, but the dual partition scheme sounds similar as in the AC1900xxx family.

stangri, 2019-01-27:
@cheneylaw -- please try luci-app-advanced-reboot build 36 from [my repo](https://stangri.github.io/openwrt-repo/) and confirm if it works.

imi2003, 2019-02-02:
@stangri 

ive tested the ipk it does reboot to the correct partition, but still shows unknown in the partition details
I have had recheck to my changes and noticed some mistakes
i was pointing to the rootfs mtd's and not the kernel

also the offset is 192

{"Linksys EA6350v3", "linksys,ea6350v3", "mtd10", "mtd12", 192, "boot_part", 1, 2},

with these changes the partition names are now shown

imi2003, 2019-02-02:
root@OpenWrt:~# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00040000 00010000 "SBL1"
mtd1: 00020000 00010000 "MIBIB"
mtd2: 00060000 00010000 "QSEE"
mtd3: 00010000 00010000 "CDT"
mtd4: 00010000 00010000 "APPSBLENV"
mtd5: 00080000 00010000 "APPSBL"
mtd6: 00010000 00010000 "ART"
mtd7: 00020000 00010000 "u_env"
mtd8: 00020000 00010000 "s_env"
mtd9: 00010000 00010000 "devinfo"
mtd10: 02800000 00020000 "kernel"
mtd11: 02500000 00020000 "rootfs"
mtd12: 02800000 00020000 "alt_kernel"
mtd13: 02500000 00020000 "alt_rootfs"
mtd14: 00100000 00020000 "sysdiag"
mtd15: 02f00000 00020000 "syscfg"

root@OpenWrt:~# dd if=/dev/mtd10 bs=1 skip=192 count=25
Linksys Linux-3.14.4325+0 records in
25+0 records outroot@OpenWrt:~# dd if=/dev/mtd12 bs=1 skip=192 count=25
OpenWrt Linux-4.14.9525+0 records in
25+0 records out
root@OpenWrt:~#

imi2003, 2019-02-02:
[image]

stangri, 2019-02-03:
Thank you for investigation and the help, I'll try to send a pr within 24 hours.

closed 2019-02-21 by hnyman