openwrt/luci #2475: Changing /etc/config/system through LuCI won't reload /etc/init.d/system by design?

frustrateddev, 2019-01-22:
Dear LuCI developers,

I've been working on LuCI web UI in a **Chaos Calmer(15.05) variant**.
Recently I've noticed that applying changes to `/etc/config/system` from LuCI won't trigger `/etc/init.d/system` to be reloaded unlike firewall or network service.
Looks like it's because the line `option init system` is not present under `config system` within `/etc/config/ucitrack`. Am I correct?

What I'm curious about here is why does LuCI do the dirty work of directly writing the updated **hostname** or **timezone** to the system files (`/proc/sys/kernel/hostname`  and `/etc/TZ` respectively) on its own, while there already exists code designed to do the same jobs and more within `/etc/init.d/system`  when it's reloaded(or started). Why doesn't LuCI simply trigger `/etc/init.d/system` to be reloaded instead? Is there a particular reason for this behavior?

==
Please take a look at the code below : 

### [luci/modules/luci-mod-system/luasrc/model/cbi/admin_system/system.lua]
```
...
o = s:taboption("general", Value, "hostname", translate("Hostname"))
o.datatype = "hostname"

function o.write(self, section, value)
	Value.write(self, section, value)
	sys.hostname(value)
end
...
```
=> Notice the code `sys.hostname(value)`, whose body includes `fs.writefile("/proc/sys/kernel/hostname", newname )`.

```
...
o = s:taboption("general", ListValue, "zonename", translate("Timezone"))
o:value("UTC")

for i, zone in ipairs(zones.TZ) do
	o:value(zone[1])
end

function o.write(self, section, value)
	local function lookup_zone(title)
		for _, zone in ipairs(zones.TZ) do
			if zone[1] == title then return zone[2] end
		end
	end

	AbstractValue.write(self, section, value)
	local timezone = lookup_zone(value) or "GMT0"
	self.map.uci:set("system", section, "timezone", timezone)
	fs.writefile("/etc/TZ", timezone .. "\n")
end
...
```
=> Notice the code `fs.writefile("/etc/TZ", timezone .. "\n")`.

### [/etc/init.d/system]
```
...
system_config() {
        local cfg="$1"

        local hostname conloglevel buffersize timezone zonename

        validate_system_section "${1}" || {
                echo "validation failed"
                return 1
        }

        echo "$hostname" > /proc/sys/kernel/hostname
        [ -z "$conloglevel" -a -z "$buffersize" ] || dmesg ${conloglevel:+-n $conloglevel} ${buffersize:+-s $buffersize}
        echo "$timezone" > /tmp/TZ
        [ -n "$zonename" ] && [ -f "/usr/share/zoneinfo/$zonename" ] && ln -s "/usr/share/zoneinfo/$zonename" /tmp/localtime

        # apply timezone to kernel
        date -k
}

reload_service() {
        config_load system
        config_foreach system_config system
}
...
```
=> Notice that the function system_config() does the jobs including what LuCI does when the service is reloaded.

jow-, 2019-01-22:
This is due to the fact that there was no standalone /etc/init.d/system in past, before https://git.openwrt.org/?p=openwrt/svn-archive/archive.git;a=commitdiff;h=5888750dabd96361bef318e693da73090abe4300 it was part of /etc/init.d/boot which wasn't side-effect free for reloads, hence the complicated logic.

The code can be probably refactored to rely on the init script logic nowadays.

frustrateddev, 2019-01-23:
I see. Thanks for the quick response. Curiosity resolved! :+1:

closed 2019-01-23 by frustrateddev