openwrt/luci #2482: IPV6 6rd protocol reports a MAC address

Cromagnon31, 2019-01-26:
IpV6 WAN 6rd protocol in the “Interfaces overview”, reports a MAC address with the value of is the “Wan” IpV4 address.
6rd protocol doesn't have MAC address in it's config.

Look at this interface overview snapshot : https://bugs.openwrt.org/index.php?getfile=643

Device used is BT Home Hub 5 v1.2.
Openwrt release is 18.06.1,
6rd package version is 9-4

adelton, 2025-01-10:
The URL with the screenshot is no longer valid, and searching the whole git history for "Interfaces overview” does not find anything.

@Cromagnon31 Is the issue still present in latest versions (23.05.5)?

closed 2025-01-10 by systemcrash

Cromagnon31, 2025-01-11:
@adelton 
I'm unable to answser. I didn't have any [Openwrt] router now.