openwrt/luci #2483: Add ordering to the DHCP screen

james-mcguire, 2019-01-26:
Ideally, the Active DHCP Leases and Active DHCPv6 Leases table should be sortable by any column, ascending or descending, defaulting to the host name, ascending..  If that's not possible, defaulting to the host name should suffice.

The Active DHCP Leases table and the Active DHCPv6 Leases table should share the same terminology (ie, both should use Host or Hostname for the first column)

These changes should apply to the Status/Overview screen as well.

Also, when assigning Static IPs on the DHCP screen, the static IP's should be sorted in ascending order.

johny-mnemonic, 2020-08-20:
I would love to see this done as well. Quite frankly OpenWrt is probably the first router interface I encountered, which is unable to sort DHCP client list. And I wouldn't be surprised if also the only one ;-)
Does anyone even use luci on OpenWrt? :-O

It is real pain to maintain static DHCP leases when you can't sort the leases at all. I am not talking just about being able to sort it as you like (which would be great!), but the lists are not sorted at all. Not by hostname, not by IP, not by lease time, just pure random :-(

There seems to be really [old ticket](https://dev.archive.openwrt.org/ticket/14355) for this already, but it somehow never got any attention, like if no one is actually using luci for dhcp management :-(

johny-mnemonic, 2020-08-20:
There are other issues reporting the same thing basically: https://github.com/openwrt/luci/issues/1370 and https://github.com/openwrt/luci/issues/1491
https://github.com/openwrt/luci/issues/1491 is even linked with PR https://github.com/openwrt/luci/pull/2623, but there seems to be the lack of interest to even merge it :-( It had been left rotting so long, that it is not mergeable now as it has conflicts.

I could try to fork it and fix those conflicts or ask the author @kctani to update it. Question is whether it has even a chance to get merged?

james-mcguire, 2020-08-20:
Hopefully!!! 😄 

I'm pretty good at resolving merge conflicts, but not very good at javascript, or else I'd take a stab at it.

ristomatti, 2020-10-05:
Couple of days ago I just installed OpenWrt for the first time in 10 years and I'm liking a lot! But the lack of sorting on any list is mind boggling and makes managing long static leases, host name maps etc. a lot slower than it would need to be...

JeanLouisCaire, 2021-02-08:
I do agree that it would be very useful to sort any table (DHCP, Hostnames, Routes etc...) by any of the columns
Do you think it would be possible to have this functionality added?

Thx

christoftaylor, 2021-07-23:
Here is javascript library that does exactly what we need and instructions on how to integrate it: 
[https://www.kryogenix.org/code/browser/sorttable/](https://www.kryogenix.org/code/browser/sorttable/ )

jow-, 2022-02-24:
Pending PR at https://github.com/openwrt/luci/pull/5696

jow-, 2022-02-25:
Fixed via #5696

closed 2022-02-25 by jow-