openwrt/luci #2489: lede-17.01, lack of description about ipv6 option

dragon100a, 2019-01-30:
Source:
downloads.openwrt.org/releases/### 17.01.6/targets/ar71xx/generic/
lede-17.01.6-ar71xx-generic-mw4530r-v1-squashfs-sysupgrade.bin

I flash this into my mw4530r. 
I try to edit the WAN interface. I meet a strange option without description or commentary.
[image]

It lose a description "Obtain IPv6-Address" and a commentary "Enable IPv6 negotiation on the PPP link".
It should be like this.

[image]

This issue occures in [lede-17.01](https://github.com/openwrt/luci/tree/lede-17.01).
For example, at 
https://github.com/openwrt/luci/blob/lede-17.01/protocols/luci-proto-ppp/luasrc/model/cbi/admin_network/proto_pppoe.lua
line 34
This line should be change to this one.
`section:taboption("advanced", ListValue, "ipv6",
		translate("Obtain IPv6-Address"),
		translate("Enable IPv6 negotiation on the PPP link"))`

This is an example. There are many pages concerned this issue.
Maybe including these folders and etc.
/protocols/luci-proto-ppp/luasrc/model/cbi/admin_network/
/modules/luci-base/po/

How to resolve this issue about missing description and commentary of lede-17.01?

jow-, 2019-01-30:
The one instance you mentioned is a code bug and has been fixed with https://github.com/openwrt/luci/commit/54455719e5f261be44e47094114dc5232492bac7. Do you have examples for the other `many pages`?

dragon100a, 2019-01-31:
> The one instance you mentioned is a code bug and has been fixed with [5445571](https://github.com/openwrt/luci/commit/54455719e5f261be44e47094114dc5232492bac7). Do you have examples for the other `many pages`?

Thanks.

https://github.com/openwrt/luci/tree/lede-17.01/protocols/luci-proto-ppp/luasrc/model/cbi/admin_network

It seems that it has been fixed.

But I'm not sure other pages. I don't know what pages will be affected by this bug.

closed 2019-02-12 by jow-

dragon100a, 2019-02-21:
It seems that my problem has been ignored.

https://github.com/openwrt/luci/blob/lede-17.01/modules/luci-base/po/en/base.po
https://github.com/openwrt/luci/blob/master/modules/luci-base/po/en/base.po

At base.po of the master tree, https://github.com/openwrt/luci/blob/c0bdb281d8a23b06bd2cb923a9e57bd24e48c776/modules/luci-base/po/en/base.po#L3430

It has "Obtain IPv6-Address“.

At base.po of lede-17, I search the whole page. And I can't find "Obtain IPv6-Address".
Does need to fix?

In addition, how to deal with my mw4530r to apply the fix above ? I need to wait the next lede edition, perhaps lede-17.01.7?
@jow-

hnyman, 2019-02-21:
> how to deal with my mw4530r to apply the fix above ? I need to wait the next lede edition, perhaps lede-17.01.7?

It is the sources and it is in the compiled packages in 17.01.

```
opkg update
opkg upgrade luci-proto-ppp
```

(But the translation files .po  have not been updated, and thus nobody has updated the translations, naturally. But that does not affect the package itself.)

hnyman, 2019-02-21:
I synced the 17.01 translations, but the only actual benefit that you will see from that, is that now you can find the untranslated string in the base.po file that you looked at earlier.

dragon100a, 2019-02-24:
> I synced the 17.01 translations, but the only actual benefit that you will see from that, is that now you can find the untranslated string in the base.po file that you looked at earlier. The updated actual package was already there since jow backported the fix in late January and the package got re-compiled.

Thanks.
This whole problem contains a code bug and all languages' translate files of luci-base(at https://github.com/openwrt/luci/tree/lede-17.01/modules/luci-base/po), not only "en" translate file(at https://github.com/openwrt/luci/blob/lede-17.01/modules/luci-base/po/en/base.po).
So I create #2559 for the "zh-cn" translate file.

To my mw4530r, 
I install the new version of luci-proto-ppp at http://192.168.1.1/cgi-bin/luci/admin/system/packages.
It does work. I finally find "Obtain IPv6-Address" and "Enable IPv6 negotiation on the PPP link" at that pages. After that, I install "luci-i18n-base-zh-cn". It lacks of Chinese translate of the pull request I mentioned above.

It looks like that all have been resolved.
To solve this, we have to upgrade "luci-proto-ppp" package, at current time. Will we have a new release of lede-17 containing above fix in future?