openwrt/luci #2492: DHCP - allow MAC clashes (same MAC for multiple subnets)

BlueMax, 2019-01-31:
Consider this case:
OpenWRT has 1xLAN (192.168.1.0/24) and 1xWiFi (192.168.200.0/24), obviously not bridged. LAN includes another router with WiFi. Only one router handles DHCP (OpenWRT) for both subnets.

A WiFi client moving from one subnet to the other still has the same MAC address but requests a new DHCP for a different subnet. So it needs multiple entries per MAC (like on the picture below) for this. Luckily, the situation below (even though marked red) does work properly nevertheless. If you press the 'save and apply' button nothing blocks the way and IPs are handed out properly regarding the corresponding subnet (as stated on the picture).

The 'Add' button is a show stopper though moaning about invalid settings and will not continue. 

The whole MAC clash situation should be allowed (at least optional) so no red area and no more moaning 'Add' buttons. :-)

[image]

OpenWrt SNAPSHOT, r9223-a5269ffa7a
luci - git-19.026.58450-d6ae363-1

jow-, 2019-01-31:
Did you check whether the resulting config would be accepted by dnsmasq?

oldium, 2020-10-22:
I think that what you want (multiple IP addresses for single client) is not allowed in [DHCP protocol](https://www.ietf.org/rfc/rfc2131.txt). There is only a single `yiaddr` field to provide IPv4 address, so what you request is impossible. If you do not want to bridge LAN and WiFi, solution would be two VLANs, one for LAN, one for WiFi. Then you need two DHCP servers to listen on interfaces corresponding to particular VLAN and on WAN router you need to add routing from both VLANs to internet (to internet you will not send VLAN tags).

BlueMax, 2020-10-22:
It does work. It's just LuCI blocking that partially.

I will update and try to bring more logs to shed some more light what really happens there.

oldium, 2020-10-22:
I am interested in how this could work on the protocol level, especially when you can offer just one IPv4 address in DHCP OFFER and the first DHCP response wins (when you try to serve multiple DHCP responses with different IP addresses).

jow-, 2020-10-22:
It does not propagate to the protocol level, it is a dnsmasq feature. You can have two or more MACs share the same static lease address which is intended for laptops or similar devices roaming between wired and wireless connections.

oldium, 2020-10-22:
I thought we are talking about one MAC with multiple addresses, not the other way around.

BlueMax, 2020-10-24:
Yes, its about one mac address, two static leases. And what can i say, it does work out of the box now (r14740). No red errors blocking in LuCI anymore and IPs get assigned appropriately. The problem is gone.

closed 2020-10-24 by BlueMax

oldium, 2020-10-25:
Most probably it works for you by luck, not by precise configuration. Forwarded DHCP REQUESTS are taken as-is, it does not matter which routers are involved in the delivery path. There is no magic, but without knowing exact configuration, this will stay mysterious to me. Good that it works for you.

BlueMax, 2020-10-26:
> Most probably it works for you by luck, not by precise configuration. 
[...]
this can break at any time).

Oh, Jedi mind tricks. Can't resist.

Not sure how i can help you here. Maybe the exact topology is of help:
Router1:
2x subnets - 1x wifi, 1x lan (provides two dhcp ranges, static leases)
Router2:
1x subnet (static ip, in subnet 'Router1/lan', dhcp disabled, bridged to wifi0).

Now, if a roaming wifi device connects via router1 it gets into 'router1/wifi'. If it connects via router2/wifi it gets into 'router1/lan'.

oldium, 2020-10-26:
> Oh, Jedi mind tricks. Can't resist.

🤣

That actually explains the situation. I will describe, what is happening.

Facts:
* Router 1 - two Dnsmasq instances, one is listening on _lan_, one is listening on _wifi_.
* Router 2 - _wifi_ is bridged to _lan_, so all DHCP requests goes to Router 1 _lan_.
* Dnsmasq help is telling this for DHCP host option: „Addresses allocated like this are not constrained to be in the range given by the --dhcp-range option, but they must be in the same subnet as some valid dhcp-range.“

Most probably your configuration in `/etc/config/dhcp` for `config host` is not using `option instance`, so all DHCP host IPs are global and used in all Dnsmasq instances. On Router 1 each instance filters the IPs based on the configured DHCP range, so the _lan_ instance accepts _lan_ IPs (and ignores the rest) and the _wifi_ instance gets _wifi_ IPs (and ignores the rest).

When you are connected to WiFi on Router 1, you get _wifi_ instance IP; when you connect to WiFi on Router 2, this is bridged to _lan_ and you get _lan_ instance's IP from Router 1.

Does it fit the actual situation?

BlueMax, 2020-10-26:
Yes, that's exactly how it is.
Thanks for clearing that up.