openwrt/luci #2503: freifunk defaults and profiles: add sections for 802.11s

pmelange, 2019-02-02:
There is general consensus in the freifunk-berlin group to move to 802.11s instead of adhoc.

@mmunz @hnyman you both have contributed to https://github.com/openwrt/luci/blob/master/contrib/package/freifunk-common/files/etc/config/freifunk.  I don't know who else to turn to in this issue.  For the move to 802.11s I would like to propose a change to the defaults, with the opportunity in the individual community profiles to override the defaults.

```
config 'defaults' '80211s'
	option 'mode' 'mesh'
	option 'encryption' 'none'
	option 'mesh_id' 'Mesh-Freifunk'
	option 'mesh_fwding' '0'
```

What would be the correct way to start getting 802.11s changes into freifunk?  Could either of you suggest additional interested people who might want to participate in the discussion?

hnyman, 2019-02-03:
No idea, I have no specific knowledge about Freifunk. 
It has mainly been @mmunz  @FreifunkUFO @pmelange , and also @jow- knows a lot.

(In general, the whole thing of Freifunk city/area wifi configs being in the LuCI GUI code repository is a strange thing nowadays, but I think that LuCI having roots with the Freifunk community plays a role there.)

SvenRoederer, 2019-02-03:
> I agree that the community profiles seem strange in LuCI. But that is another issue which we could possibly discuss separately.

See #2508

pmelange, 2019-02-24:
This issue has been moved to https://github.com/freifunk/openwrt-packages/issues/1

closed 2019-02-24 by pmelange