openwrt/luci #2508: move freifunk-related packages into a separate repo

SvenRoederer, 2019-02-03:
LuCI has strong roots in the Freifunk-Community, but nowadays LuCI has become much bigger. It feels sometimes that the existing Freifunk-packages do not really belong into the core LuCI-repo (see https://github.com/openwrt/luci/issues/2503#issuecomment-460037730).

What about moving the relevant packages into a separate repo, probably under the authority of @Freifunk?

A discussion was also started on http://lists.freifunk.net/pipermail/wlannews-freifunk.net/2019-February/004818.html

jow-, 2019-02-04:
Sure. Please prepare the separate repo and ping me when I can drop the packages from LuCI.

egmont1227, 2019-02-10:
Please find new Freifunk Global Repo at https://github.com/freifunk/openwrt-packages

SvenRoederer, 2019-02-10:
I've started a test-build of the "Freifunk Berlin Firmware" and it looks fine so far.

@jow- I'll send a PR (to delete the packages) once everything is setup.

SvenRoederer, 2019-02-13:
I just created PR# 2533 to delete the packages moved over.

SvenRoederer, 2019-02-17:
packages moved to new feed with https://github.com/openwrt/luci/commit/55cd0c4c6b572f5b35244927a738c46b7dd711b8

closed 2019-02-17 by SvenRoederer