openwrt/luci #2511: luci-mod-rpc: net.arptable - Method not found

flipreverse, 2019-02-04:
Hi all,

I want to use Home Assitant's OpenWrt plugin to track devices ([Wiki](https://www.home-assistant.io/components/device_tracker.luci/), [source code](https://github.com/home-assistant/home-assistant/blob/master/homeassistant/components/device_tracker/luci.py)).
This plugin uses luci-mod-rpc to retrieve the arp table, and uses them to track devices.
However, issuing the rpc call gives me a "method not found" error:
`
curl http://192.168.111.XXX/cgi-bin/luci/rpc/sys?auth=AUTHTOKEN --data '{"method":"net.arptable", "params":[]}'
{"id":null,"result":null,"error":{"message":"Method not found.","code":-32601}}
`
Other call like `net.ipv4_hints` work perfectly.
Does anyone know whats happening here?

Regards,
Alex

jow-, 2019-02-04:
This function has been removed from LuCI

closed 2019-02-04 by jow-

flipreverse, 2019-02-04:
Okay. Is there another way to retrieve the same information via luci-mod-rpc?

jow-, 2019-02-04:
https://github.com/home-assistant/home-assistant/pull/15792/files

mayuko2012, 2019-12-02:
I have same question with you, I find the `openwrt_luci_rpc.py` files in home-assistant deps folder has lines:
```
if self.owrt_version < version.parse("18.06"):
            return True, rpc_sys_arp_call
        else:
            return False, rpc_ip_call
```
It seems to obtained a wrong version? because I used koolshare fireware.

mludvig, 2021-08-29:
Trouble is that many OpenWRT-based firmwares use their own version scheme. I use [TurrisOS 5.2.5](https://www.turris.com/en/turris-os/) which is based on OpenWRT 19.07 but `openwrt_luci_rpc.py` thinks that the version is `5.2.5` and hence "legacy" :(