openwrt/luci #2527: [luci-app-shadowsocks-libev] Cannot disable Redir Rules

balthild, 2019-02-11:
## Step to reproduce

1. Add an remote server and ss_redir instance. Enable redir rules. In the `Local Instances` tab, the `Running` column of the corresponding ss_redir instance says `yes`；
2. Check IP (e.g. `curl https://myip.ipip.net/`) to confirm the router is forwarding connections to shadowsocks server.
3. Check the `Disabled` checkbox in the `Redir Rules` tab, then save and apply.
4. Back to the `Local Instances` tab, you can find that the `Running` column of the ss_redie instance still says `yes`.
5. Check IP and you can find that the router is still forwarding you connections.
6. Edit the local instance to disable it. Save and apply.
7. You can find that you lost your Internet access, which is likely because that the router is forwarding you connection to a stopped ss_redir instance.

## Expected behaviour

After disabling Redir Rules, the router should stop forwarding connections.

## Versions

* OpenWrt 18.06.2 r7676-cddd7b4c77
* luci-app-shadowsocks-libev git-19.034.35162-a0ae6c9-1
* shadowsocks-libev-* 3.1.3-2

balthild, 2019-02-11:
A workaround is set default behaviour to `bypass` in `Source Settings`. But it only works when there're no forward or checkdst rules presented.

yousong, 2019-02-11:
> ## Step to reproduce
> 1. Add an remote server and ss_redir instance. Enable redir rules. In the `Local Instances` tab, the `Running` column of the corresponding ss_redir instance says `yes`；
> 2. Check [...]

yousong, 2019-02-11:
> ## Step to reproduce
> 1. Add an remote server and ss_redir instance. Enable redir rules. In the `Local Instances` tab, the `Running` column of the corresponding ss_redir instance says `yes`；
> 2. Check [...]

This is expected.  "redir rules" depends on "ss-redir", and turning off "redir rules" will not affect running status of ss-redir process.

> 5. Check IP and you can find that the router is still forwarding you connections.

This is not expected.  Please try the following to further pinpoint the cause

 - check `/etc/config/shadowsocks-libev` config and confirm the ss_rules section is really disabled.  It should
 - Run `iptables-save | grep ss_rules` to see if ss_rules are still present.  They shouldn't 
 - Run /etc/init.d/shadowsocks-libev reload, then check iptable rules again

> 6. Edit the local instance to disable it. Save and apply.
> 7. You can find that you lost your Internet access, which is likely because that the router is forwarding you connection to a stopped ss_redir instance.

The init script was written to skip disabled ss-redir instance.  So this should not happen.  Anyway, things go wrong from step 5.  We should focus on that first.

Thank you for the report.

> 
> ## Expected behaviour
> After disabling Redir Rules, the router should stop forwarding connections.
> 
> ## Versions
> * OpenWrt 18.06.2 r7676-cddd7b4c77
> * luci-app-shadowsocks-libev git-19.034.35162-a0ae6c9-1
> * shadowsocks-libev-* 3.1.3-2

balthild, 2019-02-11:
> check `/etc/config/shadowsocks-libev` config and confirm the ss_rules section is really disabled.  It should

```
root@LEDE:~# cat /etc/config/shadowsocks-libev

config ss_redir 'hi'
	option local_address '0.0.0.0'
	option local_port '1100'
	option mode 'tcp_and_udp'
	option timeout '60'
	option reuse_port '1'
	option server 'cn_hk'
	option disabled 'false'

config ss_rules 'ss_rules'
	list dst_ips_forward '8.8.8.8'
	option dst_forward_recentrst '0'
	list dst_ips_bypass '202.192.18.0/24'
	list dst_ips_bypass '172.16.0.0/12'
	list dst_ips_bypass '202.192.29.0/24'
	list ifnames 'br-lan'
	option redir_tcp 'hi'
	option redir_udp 'hi'
	option dst_default 'forward'
	option local_default 'checkdst'
	option src_default 'checkdst'
	option disabled '1'

config server 'cn_hk'
	option server_port '(hidden)'
	option method '(hidden)'
	option password '(hidden)'
	option server '(hidden)'
```

> Run `iptables-save | grep ss_rules` to see if ss_rules are still present.  They shouldn't

```
root@LEDE:~# iptables-save | grep ss_rules
:ss_rules_dst - [0:0]
:ss_rules_forward - [0:0]
:ss_rules_local_out - [0:0]
:ss_rules_pre_src - [0:0]
:ss_rules_src - [0:0]
-A PREROUTING -i br-lan -p tcp -j ss_rules_pre_src
-A OUTPUT -p tcp -j ss_rules_local_out
-A ss_rules_dst -m set --match-set ss_rules_dst_bypass dst -j RETURN
-A ss_rules_dst -m set --match-set ss_rules_dst_forward dst -j ss_rules_forward
-A ss_rules_dst -m comment --comment "dst_default: forward" -j ss_rules_forward
-A ss_rules_forward -p tcp -j REDIRECT --to-ports 1100
-A ss_rules_local_out -m set --match-set ss_rules_dst_bypass_ dst -j RETURN
-A ss_rules_local_out -p tcp -m comment --comment "local_default: checkdst" -j ss_rules_dst
-A ss_rules_pre_src -m set --match-set ss_rules_dst_bypass_ dst -j RETURN
-A ss_rules_pre_src -p tcp -j ss_rules_src
-A ss_rules_src -m set --match-set ss_rules_src_bypass src -j RETURN
-A ss_rules_src -m set --match-set ss_rules_src_forward src -j ss_rules_forward
-A ss_rules_src -m set --match-set ss_rules_src_checkdst src -j ss_rules_dst
-A ss_rules_src -m comment --comment "src_default: checkdst" -j ss_rules_dst
:ss_rules_dst - [0:0]
:ss_rules_forward - [0:0]
:ss_rules_pre_src - [0:0]
:ss_rules_src - [0:0]
-A PREROUTING -i br-lan -p udp -j ss_rules_pre_src
-A ss_rules_dst -m set --match-set ss_rules_dst_bypass dst -j RETURN
-A ss_rules_dst -m set --match-set ss_rules_dst_forward dst -j ss_rules_forward
-A ss_rules_dst -m comment --comment "dst_default: forward" -j ss_rules_forward
-A ss_rules_forward -p udp -j TPROXY --on-port 1100 --on-ip 0.0.0.0 --tproxy-mark 0x1/0x1
-A ss_rules_pre_src -m set --match-set ss_rules_dst_bypass_ dst -j RETURN
-A ss_rules_pre_src -p udp -j ss_rules_src
-A ss_rules_src -m set --match-set ss_rules_src_bypass src -j RETURN
-A ss_rules_src -m set --match-set ss_rules_src_forward src -j ss_rules_forward
-A ss_rules_src -m set --match-set ss_rules_src_checkdst src -j ss_rules_dst
-A ss_rules_src -m comment --comment "src_default: checkdst" -j ss_rules_dst
```

> Run `/etc/init.d/shadowsocks-libev reload`, then check iptable rules again

```
root@LEDE:~# /etc/init.d/shadowsocks-libev reload
root@LEDE:~# iptables-save | grep ss_rules
:ss_rules_dst - [0:0]
:ss_rules_forward - [0:0]
:ss_rules_local_out - [0:0]
:ss_rules_pre_src - [0:0]
:ss_rules_src - [0:0]
-A PREROUTING -i br-lan -p tcp -j ss_rules_pre_src
-A OUTPUT -p tcp -j ss_rules_local_out
-A ss_rules_dst -m set --match-set ss_rules_dst_bypass dst -j RETURN
-A ss_rules_dst -m set --match-set ss_rules_dst_forward dst -j ss_rules_forward
-A ss_rules_dst -m comment --comment "dst_default: forward" -j ss_rules_forward
-A ss_rules_forward -p tcp -j REDIRECT --to-ports 1100
-A ss_rules_local_out -m set --match-set ss_rules_dst_bypass_ dst -j RETURN
-A ss_rules_local_out -p tcp -m comment --comment "local_default: checkdst" -j ss_rules_dst
-A ss_rules_pre_src -m set --match-set ss_rules_dst_bypass_ dst -j RETURN
-A ss_rules_pre_src -p tcp -j ss_rules_src
-A ss_rules_src -m set --match-set ss_rules_src_bypass src -j RETURN
-A ss_rules_src -m set --match-set ss_rules_src_forward src -j ss_rules_forward
-A ss_rules_src -m set --match-set ss_rules_src_checkdst src -j ss_rules_dst
-A ss_rules_src -m comment --comment "src_default: checkdst" -j ss_rules_dst
:ss_rules_dst - [0:0]
:ss_rules_forward - [0:0]
:ss_rules_pre_src - [0:0]
:ss_rules_src - [0:0]
-A PREROUTING -i br-lan -p udp -j ss_rules_pre_src
-A ss_rules_dst -m set --match-set ss_rules_dst_bypass dst -j RETURN
-A ss_rules_dst -m set --match-set ss_rules_dst_forward dst -j ss_rules_forward
-A ss_rules_dst -m comment --comment "dst_default: forward" -j ss_rules_forward
-A ss_rules_forward -p udp -j TPROXY --on-port 1100 --on-ip 0.0.0.0 --tproxy-mark 0x1/0x1
-A ss_rules_pre_src -m set --match-set ss_rules_dst_bypass_ dst -j RETURN
-A ss_rules_pre_src -p udp -j ss_rules_src
-A ss_rules_src -m set --match-set ss_rules_src_bypass src -j RETURN
-A ss_rules_src -m set --match-set ss_rules_src_forward src -j ss_rules_forward
-A ss_rules_src -m set --match-set ss_rules_src_checkdst src -j ss_rules_dst
-A ss_rules_src -m comment --comment "src_default: checkdst" -j ss_rules_dst
```

yousong, 2019-02-11:
It's bug in the init script.  Please try patching `/etc/init.d/shadowsocks-libev` like the following.  It should work.

```
From 797bc1bb6a26227ef31245730f58070765c0fde0 Mon Sep 17 00:00:00 2001
From: Yousong Zhou <yszhou4tech@gmail.com>
Date: Mon, 11 Feb 2019 13:21:04 +0000
Subject: [PATCH] shadowsocks-libev: flush ss rules on entry

---
 net/shadowsocks-libev/files/shadowsocks-libev.init | 4 +++-
 1 file changed, 3 insertions(+), 1 deletion(-)

diff --git a/net/shadowsocks-libev/files/shadowsocks-libev.init b/net/shadowsocks-libev/files/shadowsocks-libev.init
index 22d825d0b..88c1c2429 100644
--- a/net/shadowsocks-libev/files/shadowsocks-libev.init
+++ b/net/shadowsocks-libev/files/shadowsocks-libev.init
@@ -144,12 +144,14 @@ ss_rules() {
 	local args
 
 	[ -x "$bin" ] || return 1
+	"$bin" -f
+
 	config_get cfgtype "$cfg" TYPE
 	[ "$cfgtype" = ss_rules ] || return 1
 
 	eval "$(validate_ss_rules_section "$cfg" ss_validate_mklocal)"
 	validate_ss_rules_section "$cfg" || return 1
-	[ "$disabled" = 0 ] || return 1
+	[ "$disabled" = 0 ] || return 0
 
 	eval local_port_tcp="\$ss_rules_redir_tcp_$redir_tcp"
 	eval local_port_udp="\$ss_rules_redir_udp_$redir_udp"

```

balthild, 2019-02-11:
Yes, it works now. Thanks!

closed 2019-02-11 by balthild