openwrt/luci #2560: luci as user nobody timeout when calling wget

pmelange, 2019-02-24:
This is strongly related to Freifunk as the freifunk-berlin firmware is being upgraded to work with 18.06.x

The Freifunk Wizard checks upon loading LuCi if the root password has been set.  If not set, then a changePassword page is loaded.

To do this, we have an extra cgi script which looks at /etc/shadow to see if the password field has a hash and returns either yes or no.   The problem is not with the cgi script (if I run wget on the command line, the results are as expected) and has been working in the 17.01.x branch.  The problem is that the call to wget is timing out.

The code to call wget is: https://github.com/freifunk-berlin/firmware-packages/blob/4ad95b8536653c6b4c61fc358ae6c778e4d6c92d/utils/luci-app-ffwizard-berlin/luasrc/tools/freifunk/assistent/tools.lua#L68
```
local f = io.popen("wget http://localhost/cgi-bin/has_root-pass -q -O -")
local ret = f:read("*a")
...
f.close()
```

I see the call to wget in ps and it stays there for 30 seconds
```
...
4437 nobody    5388 S    {luci} /usr/bin/lua /www/cgi-bin/luci
4439 nobody    1984 S    /bin/wget http://127.0.0.1/cgi-bin/has_root-pass -q -O -
...
```

The timeout is causing every page to take 30 seconds to load.  Once the user is logged in as root, the pages load at the normal speed.

Also, a call to ```luci.sys.user.getpasswd("root")`` always returns ```nil``` when ran as nobody (password set or not).  So this doesn't seem to be an option either and is the reason we switched to a cgi script in https://github.com/freifunk-berlin/firmware-packages/commit/fe76367321b4424bb1a470e36bb2016860d38213

Have there been changes to the permission of user nobody which prevents it from using the network?  From using wget specifically?

Any help would be appreciated with the goal being that the user is first confronted with a changePassword page on a newly installed router.

pmelange, 2019-03-02:
The cause of this issue has been determined.  commit https://github.com/openwrt/openwrt/commit/c6aa9ff38870a30dbe6da17e4edad6039fe10ddf (and in 18.06 https://github.com/openwrt/openwrt/commit/e5a0b6cde051bd24280e9c3a360e65617bce2416) is preventing the cgi script from running.

feckert, 2019-04-09:
@pmelange I think this is solved so lets close them

closed 2019-04-19 by pmelange

pmelange, 2019-04-24:
fixed with https://github.com/openwrt/openwrt/commit/f00a4ae6e0b1f7b3d84e11e8dc4dd562088584e0